QIDITECH/QIDI_Q1_Pro #1: Virus warning with 4.4.15 klipper update

jleadbetter, 2024-03-27:
I'm getting a virus warning in Chrome and Windows Defender when downloading the 4.4.15 release.  To say this is worrisome is an understatement.

CChen616, 2024-03-27:
We've checked the files and they're safe. These warnings can pop up with certain types of files or newer downloads. Feel free to double-check with other antivirus software if you're still unsure.

CChen616, 2024-03-27:
Just a heads up, the update package includes components intended for use with QIDI Link, which might be flagged by antivirus software as high risk or potentially malicious. However, these components are safe and will not harm your computer. They are designed to run on the printer only after the update is installed.

xTremePower, 2024-04-05:
Hello, 
Can you explain me this ??!! It's a joke or something ? Or you really try to fool us ?

[image]

xTremePower, 2024-04-05:
Hello, can you explain me this joke ?!!

I own 4 Qidi, but in the max 3 firmwares no problem detected, in the Q1 no matter how i try, always show this
 
[image]

xTremePower, 2024-04-05:
> Just a heads up, the update package includes components intended for use with QIDI Link, which might be flagged by antivirus software as high risk or potentially malicious. However, these components [...]

Hello, 
Okay maybe, but I really have big trouble to trust someone who say me, don't wory it's a fake positive, just let it run, it's safe ...

ShootLessDie, 2024-04-06:
Hello, I am also getting this virus warning in addition to the one in the screenshot above. Please fix.
[image]

CChen616, 2024-04-08:
For those who are still concerned about the security alert, here is a temporary workaround:

1. Log into the printer's system using an SSH tool such as `MobaXterm`.
username: `root`
passwd: `makerbase`
3. Download this zip file, then use the SSH tool to upload the extracted files to the system.
[auto_update.zip](https://github.com/QIDITECH/QIDI_Q1_Pro/files/14899827/auto_update.zip)
4. Cd to where the files are and run the provided command.
```
bash auto_update.sh
```
5. After a while, you should see the screen begin to flash as the update proceeds. Wait for the update to complete.

The zip file utilizes an automatic update component introduced in version 4.4.15. After updating to this version, you can use the automatic update feature on the screen to update, bypassing the need to download the update package.

The update process requires an internet connection. For users without network access in their printing environment, it is still recommended to add the downloaded update package to the whitelist of your security software.

bgill66, 2024-05-16:
the bash script does not work when trying to run the auto update on my new Qidi Q1 Pro, current FW is V4.4.13.  Any help to either not have files blocked as virus or guidance on how to run the auto update without getting "no such file or directory" ?

greg9504, 2024-06-11:
I have version 4.4.16 firmware installed (factory).  When I run autoupdate I get "Failed to obtain online version information." Error.  Printer is connected to wifi and has access to internet.  I can not download the latest firmware because it is immediately flagged as having a virus.

CChen616, 2024-06-11:
If you can access the printer through Qidi Link but cannot update online, it is likely because the printer's system time is not synchronized, causing SSL certificate verification with the remote server to fail.

You can try using an SSH tool (I recommend Mobaxterm) to enter the printer system and run the following commands:
```bash
# username: mks
# password: makerbase
sudo apt-get remove --purge chrony
sudo systemctl start systemd-timesyncd
sudo systemctl enable systemd-timesyncd
sudo timedatectl set-ntp true
```
to synchronize the time.

Alternatively, you can add the update package to your antivirus software's whitelist.

greg9504, 2024-06-12:
Hello @CChen616,

The second command failed

CChen616, 2024-06-12:
@greg9504 
try using 
```bash
dpkg-reconfigure tzdata 
```
to set the timezone and run `sudo timedatectl set-ntp true` agian 

If it shows `NTP service: active`
```bash
mks@mkspi:/root$ sudo timedatectl set-ntp true
mks@mkspi:/root$ timedatectl
               Local time: Sat 2022-12-17 06:25:17 CST
           Universal time: Fri 2022-12-16 22:25:17 UTC
                 RTC time: Thu 2016-01-21 11:14:39
                Time zone: Asia/Shanghai (CST, +0800)
System clock synchronized: no
              NTP service: active
          RTC in local TZ: no
```

The service is already activated, just wait for a while and the time shall be synced

```bash
mks@mkspi:/root$ timedatectl
               Local time: Wed 2024-06-12 09:45:21 CST
           Universal time: Wed 2024-06-12 01:45:21 UTC
                 RTC time: Wed 2024-06-12 01:45:21
                Time zone: Asia/Shanghai (CST, +0800)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no
```

greg9504, 2024-06-12:
Thanks, but I just finished installing 4.4.19 via USB (everything is OK after installation).  I had to turn off virus protection for the folder I downloaded it.

paulmmluap, 2024-11-22:
Turn off virus protection . . . everything will be fine.  NO!  Fix your code.  This is ridiculous to ask us to turn off a threat protection.

Moto-EMT, 2024-11-22:
We need an official Firmware Variant that will allow us to disable Qidilink.  As you indicate this is the source of issues.  And we have known security vulnerabilities associated with this particular service, and even as such you have not addressed identified issues that have previously been disclosed based on last reports.  
https://github.com/snowdroppe/qidi-security-disclosure  if anyone else is unware.

zharptitza, 2024-12-23:
I think I'm cancelling my printer order after seeing this thread.

dshonz, 2024-12-29:
После обновлени постоянно ошибка 
Moonraker не смог запустить некоторые плагины, пожалуйста проверьте логи, обновите конфигурацию и перезагрузите moonraker.
file_manager
Moonraker Найдены предупреждения.
Component 'file_manager' failed to load with error: [Errno 5] Input/output error: '/home/mks/gcode_files/sda1/System Volume Information'
Конфигруцию для плагинов Moonraker можно посмотреть [здесь.](https://docs.fluidd.xyz/configuration/moonraker)

Печатать не могу... РАспишите план как сделать откат версии без последствий

Kt0-Nado, 2025-01-06:
A brand-new printer right out of the box refused to print from a USB stick.

It does print, but in the air, even though it performs calibration each time. From OrcaSlicer, it prints normally, and the built-in test models (like the boats) print fine as well. However, files from the USB stick print in the air.

I updated the firmware via USB to the latest version, but it didn’t help. Instead, the printer broke.

How did this happen?

It’s strange! I started a 6-hour print, monitored it for an hour, and everything was working perfectly. I left for work, but when I came back, I discovered that after about 4 hours, the printer had started dragging strings of filament all over the surface, got tangled in them, ripped off its own fan cover, and then froze.

The result:

The printer is 5 days old.
It has printed 2.5 kg of filament.
Now it shows a printer configuration error, a Klipper error, and does not respond to commands related to movement or heating. The only functions that work are menu navigation and changing the language.
It’s surprising that the printer has no safety system to stop it when it encounters an obstruction. It’s VERY strange that the printer pulled with such force that it ripped off half of its own head...

paulmmluap, 2025-01-10:
Orca did the same for me.  Qidi slicer works fine.  Not sure why. —Paul

EnzoneF1, 2025-03-12:
> It's hard to remove the warning, so for those who are still concerned about the security alert, here is a temporary workaround:
> 
> 1. Log into the printer's system using an SSH tool such as `MobaXterm`.
>    username: `root`
>    passwd: `makerbase`
> 2. Download this zip file, then use the SSH tool to upload the extracted files to the system.
>    [auto_update.zip](https://github.com/QIDITECH/QIDI_Q1_Pro/files/14899827/auto_update.zip)
> 3. Change working directory to where the files are using cd command, for example:
> 
> ```
> cd /home/mks
> ```
> 
> and run the provided command.
> 
> ```
> bash auto_update.sh
> ```
> 
> 4. After a while, you should see the screen begin to flash as the update proceeds. Wait for the update to complete.
> 
> The zip file utilizes an automatic update component introduced in version 4.4.15. After updating to this version, you can use the automatic update feature on the screen to update, bypassing the need to download the update package.
> 
> The update process requires an internet connection. For users without network access in their printing environment, it is still recommended to add the downloaded update package to the whitelist of your security software.

Can you help me i keep getting an error but i have a xmax 3 and i got the 3 file off of update 3.15.  my printer is stuck on the system starts abnormally but moba still works.

Traceback (most recent call last):
  File "version_check_Q1.py", line 14, in <module>
    raise Exception("Unknown model")
Exception: Unknown model
Traceback (most recent call last):
  File "download_update_Q1.py", line 25, in <module>
    raise Exception("Unknown model")
Exception: Unknown model

Kt0-Nado, 2025-11-26:
Hi. My printer broke again, for the second time this year, and it's not
pleasant. I started a print job as usual and an hour later discovered it
was printing some kind of unimaginable nest of filaments. A significant
portion of the plastic had fused into a single lump and was dripping from
the nozzle in a huge drop. Somehow, the molten plastic had even seeped up
under the print head cover. The printer wasn't responding to commands, and
printing couldn't be stopped from the menu. I unplugged the printer and
upon powering it back on, I saw a thermal sensor error message. I carefully
cleaned out everything that had spilled out and tried printing again, but
the temperature control turned out to be faulty. The nozzle heats up to 100
degrees Celsius and that's it, but the temperature reading is still 0
degrees Celsius and the error message returns. How can I fix this? As far
as I understand, the temperature sensor is broken.