QIDITECH/QIDI_Q1_Pro #2: Babystepping applies after a few lines of GCode

Pheggas, 2024-03-28:
Hi. I didn't know in which category to put this as it is not a bug or a feature. Anyway, when i was watching Maker's Muse video about Qidi Q1 Pro, he mentioned that he scrached his bed plate due to slow babystep applying.

He was basically lowering the gap too fast as the commands he sent to the printer applied after a few seconds / lines of GCODE. Here's the [video with the timestamp](https://youtu.be/6bHHX_m-F2Y?t=787) to see it yourself.

I hope this would get solved fast.

CChen616, 2024-03-28:
Currently, the printer requests babystep adjustments from Klipper using Gcode. However, due to Klipper's internal Gcode queue mechanism, there might be a delay in these adjustments taking effect. We are considering addressing this issue in the next version.

Pheggas, 2024-03-28:
Thank you. I have ordered my Q1 Pro so I will let you know if I find something inconvenient on the sw side.

trivarialthea, 2024-04-06:
Just wanted to express my gratitude to @CChen616 for the openness and responsiveness you consistently show, even for the "tinkering" user base. Your guidance on flashing a stock Klipper is truly exceptional (even though I may not use it, it is still appreciated) considering that the stock Klipper is not yet officially supported. Also, your willingness to listen to the community and address their issues by either answering them directly or forwarding them to the respective teams for discussion is just lovely.

Thank you for maintaining this approach and not letting it overwhelm you 🫶 it truly is a breath of fresh air, at least for me! 

Also sorry, yes I am aware this is outside of the topic of the issue, but since discussions are not enabled in this repository and there is no direct contact to the firmware team, I could think of no better way to express my gratitude and thank you’s. (Feel free to delete this if it does not fit)

CChen616, 2024-04-09:
Progress update: Due to the scheduling of other tasks and the need for further testing, it is anticipated that this feature will be included in V4.4.19.

Pheggas, 2024-04-09:
Alright, thank you for update!

closed 2024-05-25 by Pheggas