QIDITECH/QIDI_Q1_Pro #3: Display not working after Upgrade to 4.4.16

Groundhopper, 2024-04-04:
After the update and reboot, the dsiplay shows only - **File is too large for destination device!** 
Printing and control via Remote Qidi Slicer 1.1.2 possible.
[image]
THX for Quick Support

CChen616, 2024-04-05:
Please contact our customer support for detailed instruction on how to flash the display using SD card.
Q1support@qidi3d.com

CChen616, 2024-04-05:
If you can use ssh tools such as `MobaXterm` and still remember the printer's IP address, you can connect to it, unzip and put extracted `800_480.tft` file under `/root` folder, then restart printer.
[800_480.zip](https://github.com/QIDITECH/QIDI_Q1_Pro/files/14880587/800_480.zip)

To better identify the issue and prevent it from recurring in the future, please provide a detailed description of the steps you took to upgrade the firmware.

wizu, 2024-04-05:
Check the size of extracted file QD_Q1_UI. It should be about 14.9MB.
If it is bigger, try extracting firmware archive with different program.
I had same problem while extracting under Linux Mint. Then I've installed unrar package and used that - file size was correct.
Then use @CChen616 method to flash firmware.

Groundhopper, 2024-04-05:
> Check the size of extracted file QD_Q1_UI. It should be about 14.9MB. If it is bigger, try extracting firmware archive with different program. I had same problem while extracting under Linux Mint. [...]

@wizu i download and unzip the file with ubuntu, because windows defender gives a virus alert! So extracting is crucial! on the other hand side, the MobaXterm works, but when i'm on the printer, what is the USER and PW? i don't know...

CChen616, 2024-04-05:
@Groundhopper 
USER：`root`
PW: `makerbase`

About the virus alert, you can refer to this issue for more info
https://github.com/QIDITECH/QIDI_Q1_Pro/issues/1

wizu, 2024-04-05:
You could also rename file QD_Q1_UI to mksscreen.recovery and put it on root of your pendrive. Then stick it in printer and reboot. It should move this file in correct place and install screens update, as stated in:
https://github.com/QIDITECH/QIDI_Q1_Pro/blob/ded3ca4041c161720c1b0a97bc9550f1bc3fa731/main.cpp#L87

wizu, 2024-04-05:
@CChen616 why you are not using zip as archive method? It would be much safer option. WinRar is able to uzip files too.

Groundhopper, 2024-04-05:
@wizu @CChen616 Problem solved with the root course of CChen616 - Now update is done and Version 4.4.16 is running on the printer! THX a lot for this fast and helpful support, i really appreciate. Issue is only a matter, if your using Linux also for extracting the firmware... Noted!

CChen616, 2024-04-05:
@Groundhopper Thank you for your support of our product.

@wizu We haven't encountered issues caused by the compression format before. I will investigate the differences between the two and reconsider which format to use. Thank you for your advice.

Groundhopper, 2024-04-05:
Problem Solved be greatfull support of @CChen616 and @wizu

closed 2024-04-05 by Groundhopper