QIDITECH/QIDI_Q1_Pro #5: Add an entry to Printer.cfg

queeph, 2024-04-05:
Add the following entry to printer.cg.
`[screws_tilt_adjust]
screw1:10,7
screw1_name: Front left
screw2: 223,7
screw2_name: Front right
screw3: 122,230
screw3_name: Last right
screw_thread: CW-M4`

The bed can be better adjusted using the screws tilt adjust.
Thanks

CChen616, 2024-04-05:
We have received many similar suggestions, and we will conduct a group evaluation. If we decide to implement it, it should be included in the next version.

CChen616, 2024-04-09:
It's now included in V4.4.18

JayPee77777, 2024-04-11:
I tried the screws_tilt_calculate with the new 4.4.18 firmware.  I homed the bed first, it probed the left front, but upon probing the right front, the probe was off the edge of the print bed, and the nozzle hits the bed, and stops with an error code.  Are the probe points for these screws correct in the macro in 4.4.18 ???

CChen616, 2024-04-11:
The latest update package has been uploaded to address the issue of inaccurate positioning. 

Below are the revised settings for the screws tilt adjust:
```
[screws_tilt_adjust] 
screw1:-5,5.6
screw1_name: Front left
screw2: 216,5.6
screw2_name: Front right
screw3: 103,235.6
screw3_name: Last right
screw_thread: CW-M4
```
Due to the x axis endstop, it's not possible to achieve precise positioning for the screw in the bottom left corner.

closed 2024-07-25 by QIDITECH