QIDITECH/QIDI_Q1_Pro #6: Repository of klipper config files

wizu, 2024-04-09:
Hi,
In releases notes I could see note: Adjusted default parameters.
But I cannot find those klipper config files (eq. printer.cfg) in this repository or any other repository.
I've looked through your release changeset: https://github.com/QIDITECH/QIDI_Q1_Pro/commit/474b134e40c85b44d417eb328cea436dff7d38c5
but I cannot find anything related to screws, like statet in:
https://github.com/QIDITECH/QIDI_Q1_Pro/issues/5

I just wonder, maybe there is another repository with that changes? It would be nice to be able to see whats exactly changed in config files.

CChen616, 2024-04-10:
Config folder is now uploaded. For the future version, you can see the difference between updates.

wizu, 2024-04-10:
Great, thanks.

closed 2024-04-10 by wizu