QIDITECH/QIDI_Q1_Pro #11: Stock Klipper Flash Screen Replacement Options

samschultzponsys, 2024-04-16:
With the stock klipper flash causing loss of access to the screen, are there any known replacement screens that will allow for either a custom UI or klipper-screen to be used?

CChen616, 2024-04-16:
If you are looking for a complete screen solution, I'm currently not sure, but perhaps someone else could provide some insights. 

If you want to install Klipper Screen by yourself, you can refer to the official documentation for Klipper Screen.
https://klipperscreen.readthedocs.io/en/latest/Hardware/

leadustin, 2024-04-16:
i use this [display](https://www.elecrow.com/5-inch-qled-quantum-dot-display-800-x-480-capacitive-touch-screen-support-various-systems.html) together with a Raspberry Pi in my total [conversion guide](https://github.com/leadustin/QIDI-up2date-english)

nobrainer32, 2024-04-17:
low cost cyd display (~10€) should work too out of the box: https://youtu.be/R3o0MGYW1ZU?si=bh63JCqmxLuzlJs2

samschultzponsys, 2024-04-17:
> low cost cyd display (~10€) should work too out of the box: https://youtu.be/R3o0MGYW1ZU?si=bh63JCqmxLuzlJs2

I dont have a Q1 yet in hand to see how the stock screen attaches.

CChen616, 2024-04-18:
> I dont have a Q1 yet in hand to see how the stock screen attaches.

The stock screen uses a serial port. You can contact customer service for a schematic of the mainboard.