QIDITECH/QIDI_Q1_Pro #12: Virus detected by two scanners on firmware V4.4.18

jbkcrash, 2024-04-17:
Not trying to be negative here, and I love my new Q1 Pro, however, I just started to upgrade my Q1's firmware and as soon as I downloaded the latest release of the firmware, V4.4.18, it was isolated and it was flagged as a "virus", and not just a Malware, an actual virus. I am hoping this is a false positive, but now I am very, very reluctant to even open the .zip.

Has anyone else had this experience? It could be the nature of the contents and not an actual signature, but curious how others have either ignored or got past it. My concern is since the printer has an SBC inside it, it could be running some form of replicating virus itself.

CChen616, 2024-04-18:
Please refer to this topic: https://github.com/QIDITECH/QIDI_Q1_Pro/issues/1

jbkcrash, 2024-06-19:
Just wanted to confirm since I noticed I never came back to close this issue.

I did the upgrade, monitored the traffic to and from the printer, and agree I see no concerns.

closed 2024-06-19 by jbkcrash