QIDITECH/QIDI_Q1_Pro #13: Layer problems Q1 Pro 4.4.18

TiWe1998, 2024-04-22:
Hello team,

I have installed version 4.4.18 and unfortunately I have encountered layer problems. The Q1 Pro keeps losing its layer every now and then, making the object very buggy.

I believe that this problem has only occurred since 4.4.18. See photos.

BR Tim

[image]
[image]

CChen616, 2024-04-22:
Is the issue occurring during a normal printing process? Could you please attach your `printer.cfg` configuration file (if it has been modified), the `klipper.log` file, and the G-code file used for printing? These documents are necessary to help identify the source of the problem.

CChen616, 2024-04-23:
Please contact our customer service Q1support@qidi3d.com, they will help you identify the problem

NpFaith, 2024-04-23:
I'm having the same problem, and only notice after the last firmware 4.4.18.

TiWe1998, 2024-04-23:
@NpFaith @CChen616 

I wrote the email and will give you an update when I receive a response.

BR Tim

XxDead7sxX, 2024-04-25:
I'm also having issues since the new update, I can seem to get a consistent successful first layer.

darwin01darwin, 2024-04-25:
Same here. More often than not I get a weird mesh when the printer does the auto mesh leveling. A way to get around it for me is to abort the print and start it again. After starting again I get a better mesh somehow. This weird behavior may have to do with the screws_tilt_adjust update in the config/printer.cfg file on commit 6336e7d.[https://github.com/QIDITECH/QIDI_Q1_Pro/commit/6336e7d34e802ee8d0ca0c8012b49981cd9f63e9](url)

j-f-s, 2024-04-25:
I'd be surprised if it was that commit on the bed screws locations, Darwin. I manually made those three mods to my printer.cfg running 4.4.16 and it's not exhibiting any of the odd layer issues shown in the pictures.

darwin01darwin, 2024-04-25:
@j-f-s for me it's not a layer issue. It's more of a first layer issue that I didn't have before. It happens randomly but more often than not. I don't know what exactly the new numbers do but that's the only thing that kinda stuck out to me for the new version. Could also be that I have a faulty probe sensor.

CChen616, 2024-04-26:
It seems that the issue may be related to the material being used.

If you are printing with PETG, please reduce the speed for printing the outer shell in your slicing software to around 50 to 60 mm/s. This should significantly improve the problem.

[image]

CChen616, 2024-04-26:
We are currently trying to replicate the issue but have not yet succeeded. To expedite the identification of the cause, we kindly request that anyone experiencing similar problems provide detailed information. Please include the following:

- **Printer Configuration**: Attach your `printer.cfg` file.
- **Filament Type**: Specify the type of filament you are using.
- **Photos**: Include photos of the problematic model to illustrate the issue.
- **Log Files**: Attach relevant log files that might give insights into what went wrong.
- **Gcode Files**: Provide the Gcode files used for the print.
- **Printing Process Details**: Describe the printing process in detail, including any pauses or resumes that occurred during printing.

Not all items are mandatory, any information you can provide would be greatly appreciated and will help us resolve the issue more efficiently.

Pheggas, 2024-05-25:
Is this issue still present in 4.4.19 version?