QIDITECH/QIDI_Q1_Pro #14: Klipper UI's printing estimates shows wrong total layer count

j-f-s, 2024-04-23:
I notice lately that the total layer count in the klipper UI printing status mismatch what is shown in QIDISlicer's layer view.

[image]
[image]

The time estimate for "Finish" seems to be pretty close.

CChen616, 2024-04-23:
Please refer to this issue here: https://github.com/fluidd-core/fluidd/issues/1403

The layer count displayed on the fluidd interface is an estimate, which I believe is based on the count of movements of the Z-axis. Currently, the version of Klipper used on the Q1 does not support the `SET_PRINT_STATS_INFO` macro. We are considering adding this feature in the next version.

j-f-s, 2024-04-23:
I presume you're referring to [this commit from October 2022](https://github.com/Klipper3d/klipper/commit/b0ffb269d2ac8dc19ae6879d80718b8c60ab60de)?

CChen616, 2024-04-23:
Yes exactly, you can modify `print_stats.py` file and restart klipper to enable the macro.