QIDITECH/QIDI_Q1_Pro #17: Main screen text colour

johnruciak, 2024-04-30:
### Discussed in https://github.com/QIDITECH/QIDI_Q1_Pro/discussions/16

<div type='discussions-op-text'>

<sup>Originally posted by **johnruciak** April 30, 2024</sup>
Can we get the red highlighted text (like when the extruder is heating up) changed to yellow? Yellow would have higher contrast and be much easier to read.</div>

CChen616, 2024-04-30:
The current UI system is difficult to customize. If changes are made, they would apply to all users, hence these modifications may not be universally applicable.

However, we have taken note of this to see if there is a broader demand for this feature among users.

Deltaflyer7, 2024-05-03:
Then it would be possible to offer 2 options 1. Normal -> text color red
2. HighContrast _> text color yellow.
This way the user can choose what is better for them.

TinkerBarn, 2024-06-23:
I support @Deltaflyer7  suggestion! I have also the issue that red on black is form me difficult to read. Yellow on black would be much better.
I think it depends on my age of 55years. Eyes get worse if you get older :-)

Squeeze, 2024-10-09:
Here's another one with the age of 55.
I don't like to be a 'me too' guy, but I also have trouble reading the red text.

VirtualBlack, 2025-07-21:
I agree, the red font doesn't provide the best contrast with the grey brackground