QIDITECH/QIDI_Q1_Pro #18: Filament runout sensor logic is not very usefull

szafranski, 2024-04-30:
Hi, I hade few issues in just 2 days (frustrating) when I had to disassemble hotend, after I ran out of filament on the spool. Now it works like this:
1. Filament runs out - good
2. First sensor is not detecting this - bad
3. Sensor near printhead detects it - good
4. Printer is aware of this with DELAY - very bad (the reason is probably old Klipper versions, since newer priorities this even (like emergency stop etc.)
5. Printhead parks - good
6. Printer waits for new filament - good
7. Nozzle cools down in a meantime - terrible, because end of the filament is after the gears, but right at the beginning of the heatbreak part. There is no way to warm it and no way to pull it back. Actually it is very hard to do something about even after disassembling the hotend and removing it from the extruder. 

How it should be done (few ideas to resolve the issue):
Suggested:
1. Update the Klipper version to prioritise the runout event and stop when old filament is still between the gears, so it could be pulled back/retracted
Walkorunds for now:
1. Pause the print when first sensor detects lack of filament (a little more waste but smoother operation, at least that should be an option)
2. Not sure if that would resolve the issue, but nozzle should not cool down with old filament in it (and in the heatbreak) - at least for few minutes or so, to give time to do something 

Please, do something about it. I was forced to disassembly the hotend few times in few days. That printer is being advertised as something semi-automatic, but I was forced to smell ABS fumes for like an hour total (just last few days). Thanks

Additional info about extruder:
By the way - I think that there is too big gap between the gears and the heatbreak in this extruder. As far as I know early samples of this printer had this issue, but I am not sure what version of extruder I got. It causes troubles like this, but it also causes clogs when changing filament the "manual way" from PLA to ABS for example since PLA is stuck after gears as well (same issue as described above).

CChen616, 2024-05-02:
The first sensor you mentioned is actually a tangle detection device, not a filament runout detector. 

I have compared the 0.10.0 version of Klipper used by the Q1 model with the current 0.12.0 version again, and found no changes to the filament width sensor other than the addition of a `max_diameter` configuration option. However, we will continue to monitor this and try to reduce the time it takes from triggering to pause. Currently, our tests show that after triggering the filament detection to pause, the remaining filament in the extruder is still effective in preventing empty printing. 

Regarding the issue of maintaining heat during a pause, we did operate that way a long time ago, but we received customer feedback about potential safety risks (such as causing a fire), and issues with filament melting and clogging the extruder due to heating without extrusion over long periods. Therefore, we discontinued this feature. You can reload the filament and heat the extruder after triggering a pause, and then clean out the remaining filament by extruding before resuming printing.

szafranski, 2024-05-06:
Hi!

Thank you for the response. So:

1. I understand that first sensor is for a tangle detection, but it could also be used as a runout sensor, correct?
2. Regarding the Klipper versions response time - I meant that newer Klipper should make the print to be paused sooner, after detecting the runout - even when using current runout sensor, in the printhead. Isn't it doing this? I am not saying about differences in filament width detection. 
3. Besides - if you have working  newer Klipper on this model, will it be available officially as well? I tried using KIAUAH to update, but since this is a "dirty" version, with modified files, it is not able to make the update. I don't want to risk bricking the device as well. 
3.1 If there is a way to make the main board to boot from the USB, you can provide me this info. That would allow me to check few things, without possibility of messing the EMMC stored OS and software.
4. As for the heat during the pause, could you at least consider making nozzle not to cool down for at least few minutes? And besides - could you make the behaviour during the pause optional maybe? 

Thanks!

CChen616, 2024-05-06:
1. The working principle of this sensor is similar to a button; it only has two states: triggered and untriggered. When the filament is obstructed and cannot feed through, the spring compresses until it triggers. However, running out of filament does not compress the spring and therefore does not trigger the sensor.

2. The runout sensor is actually a filament width sensor provided by Klipper. It continuously monitors changes in the diameter of the filament until a threshold is reached, for example, 1mm, at which point a runout event is triggered and initiates several actions including pausing. In Klipper versions 0.10.0 used on the Q1 and the latest 0.12.0, there have been no significant changes to the filament width sensor, Gcode actions, or filament switch sensor components that would impact the execution efficiency of this feature.

3. Regarding updating Klipper to a new version, since there is no effective bootloader installed, firmware updates must be done manually. The Q1 extruder uses an STM32 chip, which requires specific tools and software for flashing (such as ST LINK), making the process quite complex and requiring specific knowledge. Some users may prefer not to update their firmware. If an update is released, we will need to maintain two versions of Klipper, which can be burdensome given that the Q1 is a newly launched model with frequent updates expected. I am not dismissing the possibility of updating Klipper, but we need more time to prepare everything including documentation and update packages.

    3.1 There is no capability to boot from USB :(

4. A temporary solution is to open the `printer.cfg` file and locate the `[hall_filament_width_sensor]` section. Add a command such as `M104 S200` to the end of the `runout_gcode`. For specific usable Gcode, please refer to:
https://www.klipper3d.org/G-Codes.html#g-code-commands