QIDITECH/QIDI_Q1_Pro #23: update issue 4.4.19

therealkerwalker, 2024-05-07:
I have just updated from 4.4.18 to 4.4.19 and have a UI version issue alert stating UI does not match firmware with a report of UI:18 SOC:19. Although this only appears on the Home screen and from the testing so far the printer works as normal and both fluidd (running on windows) and Qidi Link (ios) can still connect. Checked firmware update tab and all appears to be normal with no errors reported on the export file. Reset to factory default but still the same issue on the home screen. Anyone experienced this issue? Thanks

CChen616, 2024-05-07:
May I ask where you sourced the update package from? Did you download it from GitHub or use the automatic update feature? I've tested both methods and currently, there are no issues. Please try redownloading the update package and update again.

therealkerwalker, 2024-05-07:
Hi,Yes GitHub

therealkerwalker, 2024-05-07:
I did re download and re run the update procedure and the same scenario. Takes the 4.4.19 firmware, indicates its running 4.4.19 but still this UI - firmware warning on screen

CChen616, 2024-05-07:
I have verified that the files on GitHub are correct, so I suspect that the UI files have not been installed or have been overwritten by something else. Please make sure there are no other UI files or files like `mksscreen.recovery` on your USB device. After a successful update, the UI file shall be renamed to `QD_Q1_UI.bak`

Regarding the issue with connecting to the server, you can try using the automatic update feature (since the SOC has already been successfully updated, it should indicate that you are on the latest version). After that, export the logs and compress the generated QD_Log folder into a zip file for upload, which will help me analyze the issue more effectively.

therealkerwalker, 2024-05-07:
[QD_Log.zip](https://github.com/QIDITECH/QIDI_Q1_Pro/files/15234504/QD_Log.zip)
Please find attached requested.
On the update from remote sever I still get the failed to connect report. The version shows as 4.4.19 but does not show export log request for download. I only get that if I manually offline firmware update. Like I say remote has never worked right out of the box. One thing I did notice was in the QD_Update folder the file QD_Q1_UI.bak actually displays QD_Q1_UI.bak.bak.bak.bak.bak.bak, that to me seems very odd? Appreciate all the help.

therealkerwalker, 2024-05-07:
[image]
HI, Image file of the error screen for your ref. Thanks

CChen616, 2024-05-07:
The issue with the remote server is due to the printer's system time being out of sync, currently set to 2023/11/8. Our server utilizes ssl certificates for https connections, and the certificate's validity date is much later than the system time, causing the system to deem the certificate invalid.

You can update the time by navigating to the fluidd interface and finding the `time_update.cfg`. Follow the instructions there to update the time.

Alternatively, you can SSH into the printer's system and execute the following commands:
```sh
# Remove any conflicting services
sudo apt remove -y chrony ntp
# Enable NTP
sudo timedatectl set-ntp true
# Set the timezone
dpkg-reconfigure tzdata 
```

Regarding the UI file, this does indeed seem odd. Try renaming it back to `QD_Q1_UI` and only keep UI file in `QD_Update` directory then update to see if the screen turns white and enters update mode.

therealkerwalker, 2024-05-07:
Sorted. The solution was the QD_Q1_UI.bac.bac........... file. Renamed and did a manual update and its now back to normal QD_Q1_.bak after update. My guess is when its updating its referencing the old file name and adding .bak to the file name and not overwriting it adding the .bak every time as I was trying to resolve. Guess that file needs to be removed on updates. On the time sync issue I will follow the instructions later tonight and hopefully it will resolve.

But thanks for the support it is most appreciated.

closed 2024-05-07 by therealkerwalker