QIDITECH/QIDI_Q1_Pro #25: Enhancement: Start up process order

rav101, 2024-05-13:
At the moment, the start up process has the extruder going to the waste bin, cleaning the nozzle, then going to the front to heat up the filament which is the drops some onto the print bed before starting the print.

Two potential enhancements:

1) Depending on the material used, cleaning the nozzle might not be required. Potentially consider adding this as an option to turn on / off so users can decide if they need it for their given material

2) If you are going to deposit the initial heat up material, wouldn't it make sense to do this in the waste bin as well rather than onto the print bed? Maybe there is a reason the extruder is moved to the front but time could also be saved if it didn't have to move to the front.

There are potentially other improvements in the start up process but these two steps would keep for a cleaner build environment and also reduce some of the time for printing.

CChen616, 2024-05-14:
When the print head moves to the front during preheating, it's because the printer usually prints a purge line before starting the print, which is typically located at the bottom left corner.

We will consider your suggestion. In the meantime, you can achieve these features now by modifying the following section in the `gcode_macro.cfg` file:
```
[gcode_macro PRINT_START]
gcode:
    AUTOTUNE_SHAPERS

    {% set bedtemp = params.BED|int %}
    {% set hotendtemp = params.HOTEND|int %}
    {% set chambertemp = params.CHAMBER|default(0)|int %}
    set_zoffset
    M104 S0
    M140 S{bedtemp}    
    M141 S{chambertemp}
    G28    
    CLEAR_NOZZLE HOTEND={hotendtemp}

    M190 S{bedtemp}   
#    Z_TILT_ADJUST
    G29
    G0 Z50 F600
    G0 X0 Y0  F6000
    
    M191 S{chambertemp}
    M109 S{hotendtemp}
    M204 S10000

    ENABLE_ALL_SENSOR
    save_last_file
```
For example, you can skip the nozzle cleaning by commenting out `CLEAR_NOZZLE HOTEND={hotendtemp}`, or adjust the preheat position by modifying the location of the lines:
```
    G29
    G0 Z50 F600
    G0 X0 Y0  F6000
```

j-f-s, 2024-05-14:
The way I dealt with the default purge line dragging ooze over into my print (which it did 50% of the time) was to use the same one I set up for my Prusa and Voron that purges at the origin (front left) corner and also does a thick purge line. It does a few things I find useful:

1. Since it starts the purge line from the outside edge of the print bed, any ooze debris stays over there usually hanging off the side of the build plate out of the way of the print. Also usually helps "pluck off" any leftover ooze from the nozzle tip.
2. Since the last 2/3 of the purge line is pretty thick, it puts a little flow stress test on the extruder. If the thicker part of the purge line isn't nice and smooth and shiny, you probably have a temperature issue, an extruder issue, or some other filament melt/extrusion problem. You can actually see bubbles in wet filament in this purge line pretty easily.
3. Since it's thicker, it is WAY easier to remove it, especially on textured plates, and even more especially when your z-offset or extrusion might be slightly off. 

```PRINT_START BED=[first_layer_bed_temperature] HOTEND=[first_layer_temperature] CHAMBER=[volume_temperature]
M83
G4 P3000
;##### comment out all the G0/G1 stuff from the original L-shaped purge line right next to the print
;##### COPY over my prusaslicer purge line setup.
G1 X0 Y0.0 F1000.0 ; go to origin corner of print area
G92 E0.0
G1 Z0.5 F1000.0 
G1 X30.0 E9.0 F1000.0 ; intro line
G1 X50.0 E12.5 F1000.0 ; intro line, thicker
G92 E0.0```

Hope someone finds it helpful.

rav101, 2024-05-17:
Thanks all,

I will test out these settings. I'm currently waiting for the printer to arrive and will test and tweak to figure out what works for me.

closed 2024-05-17 by rav101

robotm-team, 2024-05-30:
@j-f-s Welcome.
This is (will be) my first contact with a clipper so it will probably sound silly, but:
the block you specified should be entered into the slicer as ‘g-code start’ or ‘somewhere’ in the printer configuration ?

j-f-s, 2024-05-30:
> @j-f-s Welcome. This is (will be) my first contact with a klipper so it will probably sound silly, but: the block you specified should be entered into the slicer as ‘g-code start’ or ‘somewhere’ in the printer configuration ?

Printer Settings -> Custom G-code -> Start G-code block.

robotm-team, 2024-05-31:
Is that right or have I commented too much ?

```
PRINT_START BED=[first_layer_bed_temperature] HOTEND=[first_layer_temperature] CHAMBER=[chamber_temperature]
M83
G4 P3000
# G0 X{max((min(print_bed_max[0], first_layer_print_min[0] + 80) - 85),0)} Y{max((min(print_bed_max[1], first_layer_print_min[1] + 80) - 85),0)} Z5 F6000
# G0 Z[first_layer_height] F600
# G1 E3 F1800
# G1 X{(min(print_bed_max[0], first_layer_print_min[0] + 80))} E{85 * 0.5 * first_layer_height * nozzle_diameter[0]} F3000
# G1 Y{max((min(print_bed_max[1], first_layer_print_min[1] + 80) - 85),0) + 2} E{2 * 0.5 * first_layer_height * nozzle_diameter[0]} F3000
# G1 X{max((min(print_bed_max[0], first_layer_print_min[0] + 80) - 85),0)} E{85 * 0.5 * first_layer_height * nozzle_diameter[0]} F3000
# G1 Y{max((min(print_bed_max[1], first_layer_print_min[1] + 80) - 85),0) + 85} E{83 * 0.5 * first_layer_height * nozzle_diameter[0]} F3000
# G1 X{max((min(print_bed_max[0], first_layer_print_min[0] + 80) - 85),0) + 2} E{2 * 0.5 * first_layer_height * nozzle_diameter[0]} F3000
# G1 Y{max((min(print_bed_max[1], first_layer_print_min[1] + 80) - 85),0) + 3} E{82 * 0.5 * first_layer_height * nozzle_diameter[0]} F3000
# G1 X{max((min(print_bed_max[0], first_layer_print_min[0] + 80) - 85),0) + 3} Z0
# G1 X{max((min(print_bed_max[0], first_layer_print_min[0] + 80) - 85),0) + 6}
# G1 Z1 F600
G1 X0 Y0.0 F1000.0 ; go to origin corner of print area
G92 E0.0
G1 Z0.5 F1000.0 
G1 X30.0 E9.0 F1000.0 ; intro line
G1 X50.0 E12.5 F1000.0 ; intro line, thicker
G92 E0.0
```

j-f-s, 2024-06-01:
Yours is the same as mine, now. Hopefully you have good luck with it as I have. ;)