QIDITECH/QIDI_Q1_Pro #26: Manual filament change gcode macro

AnthonyH-22, 2024-05-16:
I want to do manual filament changes during prints. I use OrcaSlicer and when I put a filament change in on the slicer preview it just keeps printing. Orca sends a M600 when adding the filament change in the slicer preview. The M600 macro is missing from the gcode_macro.cfg file. Can that be added in a future update? 
In the interim could I add the macro for the manual filament change from the Klipper repo shown below? Would that work, or does it need to be modified for the Q1 Pro?

https://github.com/Klipper3d/klipper/blob/master/config/sample-macros.cfg
######################################################################
# Filament Change
######################################################################

# M600: Filament Change. This macro will pause the printer, move the
# tool to the change position, and retract the filament 50mm. Adjust
# the retraction settings for your own extruder. After filament has
# been changed, the print can be resumed from its previous position
# with the "RESUME" gcode.

[pause_resume]

[gcode_macro M600]
gcode:
    {% set X = params.X|default(50)|float %}
    {% set Y = params.Y|default(0)|float %}
    {% set Z = params.Z|default(10)|float %}
    SAVE_GCODE_STATE NAME=M600_state
    PAUSE
    G91
    G1 E-.8 F2700
    G1 Z{Z}
    G90
    G1 X{X} Y{Y} F3000
    G91
    G1 E-50 F1000
    RESTORE_GCODE_STATE NAME=M600_state

PPAC37, 2024-05-16:
Hello !
Personally, I changed the "filament change g-code" under my slicer to “PAUSE”.

I don't know if the M600 macro that you are offering is functional or not. But it seems that it is better to cut the old filament at the entrance of the extruder, and to extrude to purge with the new filament rather than to retract/eject the old filament in order to avoid the risk of end up with a small piece of filament stuck in the extruder. (This has happened to me before, so I now always prefer to extrude and never eject.)

AnthonyH-22, 2024-05-16:
Thanks for your reply! I have also used the Pause command (M0) in the change filament g-code. I prefer to retract the filament rather than cut and purge. Less mess and waste. So far I have not had any problems with that. I was thinking it would be nice to have it automatically retract while the extruder is hot.

CChen616, 2024-05-16:
We already have the unload gcode here:
```
[gcode_macro M603]
description: filament unload
gcode:
    M118 Heat up complete
    G92 E0
    G0  E15 F400
    G4  P1000
    G92 E0
    G1  E-90 F800
    M400
    M118 Unload finish
```
If you prefer retract directly instead of extrude then retract, you can delete `G0  E15 F400` , but be ware of the risk of clogging.

Combined with pause it would be something like this:
```
[gcode_macro M600]
gcode:
    SAVE_GCODE_STATE NAME=M600_state
    PAUSE
    M109 S220
    M603
    RESTORE_GCODE_STATE NAME=M600_state
```

AnthonyH-22, 2024-05-16:
Thanks for the quick reply. I will give the custom M600 a try and see how it works out.

AnthonyH-22, 2024-05-16:
It works great! Thank you! For anyone that wants to do easy filament changes add this to the gcode_macro.cfg and then on the printer settings in the slicer add M600 in the Change Filament G-code section. It will pause at the layer you put the filament change on and park extruder at the poop box and unload the filament. You then insert the new filament and press Load on the front screen and it will load it. Then press Play and it will pick up where it left off in the print.

[gcode_macro M600]
gcode:
    SAVE_GCODE_STATE NAME=M600_state
    PAUSE
    M109 S220
    M603
    RESTORE_GCODE_STATE NAME=M600_state

dewi-ny-je, 2024-06-02:
Since macros are lost at every update, those commands can be placed maybe in the slicer settings for change fiament GCODE.

Merlin8000, 2024-09-04:
Is it possible for this to be included in a future firmware release?

dewi-ny-je, 2024-09-05:
> Is it possible for this to be included in a future firmware release?

I'm not sure it's needed.
In Orca in the "change filament gcode" I simply put "PAUSE" and it works also like that: since manual interaction is needed, I swap filament using the display buttons.

So this macro is completely optional and the printer does NOT need to provide a M600 to be able to swap filament.

M600 is needed if automatic swaps are possible. It's not the case for Q1.

KazakovAndrey, 2024-12-27:
look here https://wiki.qidi3d.com/en/software/qidi-slicer/print-settings/insert-pause-gcode
It doesn't work!

AndreaPro, 2024-12-27:
> look here https://wiki.qidi3d.com/en/software/qidi-slicer/print-settings/insert-pause-gcode It doesn't work!

Can you explain what "It doesn't work!"?