QIDITECH/QIDI_Q1_Pro #27: Release 4.4.19 moving target?

derFliegendeHamburger, 2024-05-16:
The zip file for this release from May 04th has been replaced at least once (now from May, 15th) but no info was supplied.
What has changed or been fixed? Should it be installed again over older 4.4.19 (if at all possilble)?

BTW, the provided sources are outdated from mid April.

Please, numbers are cheap; if there was a change then increment the version number

CChen616, 2024-05-17:
Changes include optimizations for auto-updating (now downloading from your selected server) and adjustments to the `printer.cfg` and `gcode_macro.cfg` files to address issues with downward homing failures reported by some users.

In general, for users who have already updated and are not experiencing issues, there is no need to update again. However, if you wish to update, you can do so by downloading the update package and performing an offline update.

We acknowledge that our previous approach was unprofessional and we will avoid similar situations in the future.

I will upload the updated content and explanations to Git shortly.

derFliegendeHamburger, 2024-05-17:
sounds perfect.
Thank you for understanding and for the quick resolve

closed 2024-05-17 by derFliegendeHamburger

rav101, 2024-05-17:
Just a comment to say thank you to @CChen616 for taking the feedback and acting in it so quickly!

For those of us who value transparency and open source-ness, it is really appreciated!