QIDITECH/QIDI_Q1_Pro #28: Printer Doesn't Report Tangle to Fluidd / Remote Interface

smurfhunter, 2024-05-22:
Noticed a likely oversight today while I was out of my house - the touch screen will report a jam and pause the printer, but there's no output to fluidd (not even a console note) as to why the printer paused - it will just pause itself. This was initially confusing to me at first because I thought I hit pause on accident and had to interpret that the roll might have tangled.

I imagine this would be something that's not too hard to do - just having a line go to the console to indicate to the user it's tangled. I don't know if there's actual UI to indicate a tangle in fluidd, but output to the console would be a lot better than only pausing.

CChen616, 2024-05-22:
Open the printer.cfg file. Locate the section:
```
[filament_switch_sensor fila]
pause_on_runout: True
runout_gcode:
            PAUSE
event_delay: 3.0
pause_delay: 0.5
switch_pin:U_1:PC3
```
Modify the runout_gcode option to:
```
runout_gcode:
            M118 Filament tangle detected
```

This way, when a filament tangle is detected, the Fluidd interface will display a corresponding message.

We will include this change in the default settings in the next version.

smurfhunter, 2024-12-19:
Issue has been resolved

closed 2024-12-19 by smurfhunter