QIDITECH/QIDI_Q1_Pro #29: [BUG] Stopping after starting the job won't work

Pheggas, 2024-05-24:
Hey. A few days ago I received my Q1 Pro. I have firmware V4.4.16 and noticed when I started a print job and wanted to stop it a few seconds after, it would just say "Stop processing, please wait..." and the job would continue with all the macros before actual printing.
[image]

CChen616, 2024-05-25:
This issue should have been fixed in version 4.4.19. Please try updating to the latest version to see if the problem is resolved.

Pheggas, 2024-05-25:
Right. I was reading notes to each version but didn't know this bug is related to that fix. Thank you and I will update the firmware. If the issue wouldn't be present anymore in the latest firmware, I will close this issue with comment.

You can ignore this issue for now.

closed 2024-07-25 by QIDITECH