QIDITECH/QIDI_Q1_Pro #30: Mainboard and toolhead board pinouts

53Aries, 2024-05-28:
Any chance we can have the pinout diagram and/or schematic for both the main board and toolhead board? Thanks

CChen616, 2024-05-28:
Here you go
[A-8(RP) V1.0_001 PIN.pdf](https://github.com/QIDITECH/QIDI_Q1_Pro/files/15463427/A-8.RP.V1.0_001.PIN.pdf)
[X-7 V1.1_001 PIN.pdf](https://github.com/QIDITECH/QIDI_Q1_Pro/files/15463428/X-7.V1.1_001.PIN.pdf)

53Aries, 2024-05-31:
My unit has A-7 toolhead board. Do you have a pinout for that?

> Here you go [A-8(RP) V1.0_001 PIN.pdf](https://github.com/QIDITECH/QIDI_Q1_Pro/files/15463427/A-8.RP.V1.0_001.PIN.pdf) [X-7 V1.1_001 PIN.pdf](https://github.com/QIDITECH/QIDI_Q1_Pro/files/15463428/X-7.V1.1_001.PIN.pdf)

CChen616, 2024-06-01:
[A-7 V1.0_002 PIN.pdf](https://github.com/user-attachments/files/15519717/A-7.V1.0_002.PIN.pdf)

53Aries, 2024-06-01:
Can you tell me if on the Q1 the A-7 toolhead board is using CAN or USB?

CChen616, 2024-06-01:
It's using serial, and here is the make menuconfig for the toolhead, it might require using STM32-Link to flash

[image]

53Aries, 2024-06-01:
> It's using serial, and here is the make menuconfig for the toolhead, it might require using STM32-Link to flash [...]

I believe this is for the mainboard. The toolhead is an rp2040.

CChen616, 2024-06-01:
Ah, yes, I confused it with another model

53Aries, 2024-06-01:
> Yes, I confused it with another model

No worries. I ask because I'm unable to find the rp2040 toolhead board listed in the linux environment. It doesn't show as a usb or serial device. Wondering how the x7 mainboard host is configured to connect with the A7 toolhead. In the qidi config the toolhead mcu is listed as ttyS2 so i assume serial.

Wuehli, 2024-06-07:
Hi, 
is there also a pinout for the X-7 v1.0, as it seems to be different from the 1.1 you shared above.

Additional question, is there a possibility to bypass the SOC and connect e.g. a Raspberry to the STM32F402 MCU?
Thanks!!

CChen616, 2024-06-08:
[X-7 V1.0_001 PIN.pdf](https://github.com/user-attachments/files/15746098/X-7.V1.0_001.PIN.pdf)

> Additional question, is there a possibility to bypass the SOC and connect e.g. a Raspberry to the STM32F402 MCU?
> Thanks!!

Not sure 😢

Seviged, 2024-08-29:
> Hi, is there also a pinout for the X-7 v1.0, as it seems to be different from the 1.1 you shared above.
> 
> Additional question, is there a possibility to bypass the SOC and connect e.g. a Raspberry to the STM32F402 MCU? Thanks!!

Technically, it is possible. The STM32F401 uses PA11 (D-) and PA12 (D+) for communication through the USB interface. The X7 v1.0 board has a m-USB connector wired to PA11 and PA12, but the connector is not soldered, and it may lack the 22-30 ohm resistors (I don't have this motherboard yet, and the X7 doesn’t have any schematics to confirm my assumptions).

If it is possible to solder the missing components on this motherboard, you can flash Klipper (or better yet, [Katapult](https://github.com/Arksine/katapult)) to the STM32 using an SD card. In the menuconfig, choose USB (PA11, PA12). After flashing Katapult via the SD card, you can flash Klipper without the SD card directly from the SSH console.

The X7 v1.1 doesn't have a micro-USB connector onboard, but the PA11 and PA12 pins are still present at the FAN4 and FAN5 connectors. In this case, you might need to desolder some components and build the USB connection "from scratch."

All the above steps might be necessary if you want to connect via USB, essentially to a main PC or home server with an AMD64 architecture.

Another option is to use the PA09 and PA10 pins (USART1). According to the pinout, these lines have 100-ohm resistors. You could desolder them, attach some wires to the MCU (STM32) side, and connect it to another UART 3.3V-compatible interface through new 100-ohm resistors. This option is mainly for ARM64 devices (like the Raspberry Pi). You can also connect it to a USB-UART converter (preferably one with an original FTDI chip).

All of the above is theoretical; I haven't tried it yet. I’m a bit disappointed that Qidi didn’t mention the possibilities with this motherboard and seems to have cut off the MCU USB interface in the 1.1 version.

53Aries, 2025-01-04:
> > Hi, is there also a pinout for the X-7 v1.0, as it seems to be different from the 1.1 you shared above.
> > Additional question, is there a possibility to bypass the SOC and connect e.g. a Raspberry to the STM32F402 MCU? Thanks!!
> 
> Technically, it is possible. The STM32F402 uses PA11 (D-) and PA12 (D+) for communication through the USB interface. The X7 v1.0 board has a m-USB connector wired to PA11 and PA12, but the connector is not soldered, and it may lack the 22-30 ohm resistors (I don't have this motherboard yet, and the X7 doesn’t have any schematics to confirm my assumptions).
> 
> If it is possible to solder the missing components on this motherboard, you can flash Klipper (or better yet, [Katapult](https://github.com/Arksine/katapult)) to the STM32 using an SD card. In the menuconfig, choose USB (PA11, PA12). After flashing Katapult via the SD card, you can flash Klipper without the SD card directly from the SSH console.
> 
> The X7 v1.1 doesn't have a micro-USB connector onboard, but the PA11 and PA12 pins are still present at the FAN4 and FAN5 connectors. In this case, you might need to desolder some components and build the USB connection "from scratch."
> 
> All the above steps might be necessary if you want to connect via USB, essentially to a main PC or home server with an AMD64 architecture.
> 
> Another option is to use the PA09 and PA10 pins (USART1). According to the pinout, these lines have 100-ohm resistors. You could desolder them, attach some wires to the MCU (STM32) side, and connect it to another UART 3.3V-compatible interface through new 100-ohm resistors. This option is mainly for ARM64 devices (like the Raspberry Pi). You can also connect it to a USB-UART converter (preferably one with an original FTDI chip).
> 
> All of the above is theoretical; I haven't tried it yet. I’m a bit disappointed that Qidi didn’t mention the possibilities with this motherboard and seems to have cut off the MCU USB interface in the 1.1 version.

On this topic, I was poking around and found the pads on the rear of the board where PA11 and 12 would have the 22ohm resistors and confirmed they are connected at the MCU. Looking at the stm32 data sheet I don't know that having the resistors is necessary, but that's above my head right now. My question is what about ground and shield? The upper portion of the smaller pads on the left and right of PA11/12 are connected to ground. The larger center pads are PA11/12. @CChen616 could we pleeeeeease get the schematic for x-7 1.0? It's old enough at this point that surely @QIDITECH wont sustain any loss from sharing it. 
[image]

53Aries, 2025-01-04:
I have confirmed you can connect to the PA11/12 pads on the rear of the board with D+ and D- of a usb cable, ground, flash the stm with klipper built for USB communication and it works. 
[image]
[image]

Bytedecode72, 2025-06-19:
does anyone know if there are full circuit board schematics for the adapter plates A7-v1.0 (q1 pro) that when the front cover fell off it snapped the cables to the cooling fan and they shorted out, the fan survived but there is now no power on the green socket on the adapter plate so have ordered another one but would like to fix the original and odds are it will be one maybe two componets that died as a result if not i see tracing and alot of testing to do

dewi-ny-je, 2026-09-12:
> Yes, I confused it with another model

so which USART pins are the correct ones for the Q1 toolhead? UART0 on GPIO0/1?

Thanks