QIDITECH/QIDI_Q1_Pro #32: Emergency Stop

Pukkahq, 2024-05-29:
When printing the is no way to emergency stop the machine without going into the web browser. (I found out the hard way ran the bed into the print nozzle) pressing the stop button takes a few seconds to work.

Are they any extra inputs on the mainboard that we could wire one in or can one be added to the touch screen?

At the moment on the homepage we get an emergency stop
[image]

When printing, (could the pause/stop be moved under the 4 squares and add emergency stop on the screen?)

[image]

CChen616, 2024-05-30:
In our previous design, placing the button on the printing interface posed a risk of accidental touches. Such a touch would result in the entire print being ruined. Generally, collisions during printing are unlikely when using G-code files generated by slicing software, except in cases of incorrect Z-offsets or erroneous G-code macros.

Pukkahq, 2024-05-30:
Thanks you for coming back with a detialed answer. could you add a confirmation message yes/no or put it under the 4 square?

Is it possible to add a hardware button? are they any spare IO on the mainboard?

CChen616, 2024-05-30:
I will pass your suggestion to the team. However, personally, I think such an emergency stop button can be quite cumbersome. When I encounter a printing issue and need to stop urgently, I usually just turn off the power switch (which might not be a good habit). 

Currently, there are no extra IO ports on the mainboard.

Pukkahq, 2024-05-30:
Where the printer is installed the power button is not quick to get at. I spose an option would be a wireless socket and remote.

On another question is they a correct way to shutdown the printer or just switch off at the power switch?

CChen616, 2024-05-30:
Turning off the power directly should be sufficient. We have always done it this way and haven't encountered any issues so far (our power supply has a protection mechanism).

closed 2024-06-01 by Pukkahq