QIDITECH/QIDI_Q1_Pro #33: Please add option to delete files on the Qidi Q1 Pro printer screen interface

Vectorqidi, 2024-06-07:
It would be great if you could add the option to delete files of prints on the Qidi Q1 Pro screen. Right now there is no on screen option you must go through the Fluidd software.

CChen616, 2024-06-07:
This could be considered as a future improvement, but it might not be a high priority due to its limited use cases. Currently, when uploading files, you typically use slicing software or the Fluidd interface, and in both scenarios, you can generally access Fluidd to delete files.

Vectorqidi, 2024-06-07:
OK thanks, but there are a lot of people who still use a USB to transfer files and not over wifi or network so I think it would be helpful. Thank you for your response

Vectorqidi, 2024-06-11:
Incorrect.  When you print from USB the file transfer to the local memory of the printer and the small screen makes it take a long time to navigate thru the files. The ui is pretty slow and laggy.... So this is why a delete file on the printer itself option  would be useful.

robotm-team, 2024-06-15:
If it were possible to change the display mode from thumbnails to name, it would be much faster.
At the moment you have to wait for the thumbnails to load onto the screen, and it takes ages.
And if you enter the menu by accident, you can go mad.

dewi-ny-je, 2024-06-15:
I also have to say that to me thumbnails are of no use to me as well and a faster list display would be good enough, or a list view with much smaller (and faster) thumbnails would be better

Vectorqidi, 2024-06-15:
Yes the printer screen is very slow responding even when just selecting things, or changing a setting. But you are right... very very slow loading Thumbnails sometimes takes about 20-30seconds or so just for them to load. The printer prints well but the user experience is not great. The screen is so slow it makes the user experience bad. In this day and age that should not be that way. Even cheap printers these days don't have slow lagging screens so not sure what they were thinking on this printer.