QIDITECH/QIDI_Q1_Pro #39: Exhaust fan activate at complete print inquiry

Ross922, 2024-06-10:
Is there a different method to ensuring the Q1 Pro activates the Exhaust Fan after a print with ABS/ASA? 

When I set up the following snapshot below at the completion of a print the exhaust fan will activate for 2-3secs and then turn off and the printer goes into low power mode a few minutes later (LED's off, display off, etc).

Thank You

[image]

derFliegendeHamburger, 2024-06-13:
can you look at the g-code from your slicer to find whether it contains near the very end the command to activate this fan @50%?
Even if I manually add M106 S3 P122 to the end g-code section of the printer in the slicer, the fan does not run.

Ross922, 2024-06-13:
> can you look at the g-code from your slicer to find whether it contains near the very end the command to activate this fan @50%? Even if I manually add M106 S3 P122 to the end g-code section of the printer in the slicer, the fan does not run.

M141 S0
M104 S0
M140 S0
G1 E-3 F1800
G91
G0 Z5 F600
G90
G0 X0 Y0 F12000
{if max_layer_z < max_print_height / 2}G1 Z{max_print_height / 2 + 10} F600{else}G1 Z{min(max_print_height, max_layer_z + 3)}{endif}

derFliegendeHamburger, 2024-06-13:
what you posted is also the default end-g-code from within orca's definition of the printer.
You can try two things:
a) look a bit further up in the g-code if you find M106 command
b) add the M106 command at the end of the end-g-code for orca's definition of the printer and test if it gets executed.

Try this command from slicer's console window first

Ross922, 2024-06-14:
> what you posted is also the default end-g-code from within orca's definition of the printer. You can try two things: a) look a bit further up in the g-code if you find M106 command b) add the M106 command at the end of the end-g-code for orca's definition of the printer and test if it gets executed.
> 
> Try this command from slicer's console window first to verify it does what you want

Hello and thank you, would you happen to know how to set a gcode timer? i.e. 15mins and turn off?

M106 P3 S128 turns the fan on to 50% but id like it to only run for 15mins max.

CChen616, 2024-06-14:
In the `gcode_macro.cfg` file, there is a `PRINT_END` macro that executes after the print is completed. The settings from the slicer are added only at the end of the gcode file, while print_end is executed after the gcode file finishes, making its execution order later.

If you want to keep the fan running when the print ends, you can change `M106 P3 S0` to `M106 P3 S127`.

If you want to add a timer, you can add the following to the end of the macro:
```
G4 P900000
M106 P3 S0
```

After modifying the configuration, restart Klipper to apply the changes.

derFliegendeHamburger, 2024-06-19:
Your solution from above just does what I wanted. Thank you.
I think it is fair to close this issue as solved

closed 2024-06-19 by Ross922