QIDITECH/QIDI_Q1_Pro #40: Camera stopped working after security updates

LuciusDante, 2024-06-11:
Hi after I make security updates via `sudo apt upgrade` camera stopped working,
 any idea to make it work again

[image]

[webcamd.log](https://github.com/user-attachments/files/15794066/webcamd.log)

CChen616, 2024-06-12:
Can't determine the exact cause of the issue. One possible solution is:

Open the camera settings in Fluidd.
If it shows
[image]

Change the URL to

`/webcam/?action=stream`

LuciusDante, 2024-06-12:
It is same as you just default settings, what I discovered after updating packages it messed up with my mjpg streamer libjpeg8 dependicies so I ordered mks emmc to flash original sw. Anyone read this please dont use sudo apt-get update and upgrade commands.

LuciusDante, 2024-06-27:
Late feedback...
I ve tried multiple ways  rewind camera setings and drivers but none of them worked only way is flashing emmc and using recovery methode to flash ui because after flashing emmc display ui will not work due version inconstancy so you have to use recovery method mentioned in qidi wiki page also make sure plug usb when machine is on then turn off and wait 30 sec then turn on again because instructions are not clear in the wiki, I discovered this myself.

closed 2024-06-27 by LuciusDante

flaviut, 2024-07-07:
The camera is controlled by `/etc/systemd/system/webcamd.service`, which is not included in the upgrade package, but really should be. The logs for this are placed in `/var/log/webcamd.log`, so it's not easy to find (IMO it'd be better & easier to just put them in the system journal).

Anyway, the problem here is that the webcam is hardcoded as `-d /dev/v4l/webcam0` in the `/home/mks/klipper_config/webcam.txt`, but the updated packages use a more predictable naming scheme. Updating the config like this resolves the issue:

```diff
-camera_usb_options="-r 640x480 -f 10 -d /dev/v4l/webcam0"
+camera_usb_options="-r 640x480 -f 10 -d /dev/v4l/by-id/usb-SYX-231020-J_HD_Camera-video-index0"
```