QIDITECH/QIDI_Q1_Pro #41: extruder fan is pulsing after print with temp around 50degC

derFliegendeHamburger, 2024-06-13:
when print is finished, extruder fan keeps running to cool printhead down, ok.
But when extruder temp is down to around 51degC, then it begins to pulse on/off repeatedly until extruder temp is down to below 49degC. This effect may be worse if chamber and heater bed are at temps of around or above same trigger temp.

I understand the need for fan to cool down the printhead. But repeated on/off pulsing of the small fan is annoying.
If possilbe, please change PIDs or logic to avoid frequent pulsing on/off.

CChen616, 2024-06-15:
You can replace the `/home/mks/klipper/klippy/heater_fan.py` file with the one below and restart the machine to achieve the desired effect. We may not integrate this feature into the official release.

Relevant explanations have already been written into the comments in py file.

[heater_fan.zip](https://github.com/user-attachments/files/15845379/heater_fan.zip)

derFliegendeHamburger, 2024-06-18:
i found the file to replace  in _/home/mks/klipper/klippy/extras_ and after machine restart the corresponding .pyc file was updated.
Now when all thermals are cooling down, the heater fan stops when extruder temp drops to 54degC with no repeated on/off pulsing.
Just what I hoped for. Thank you

derFliegendeHamburger, 2024-06-19:
Your solution from above just does what I wanted. Thank you

closed 2024-06-19 by derFliegendeHamburger