QIDITECH/QIDI_Q1_Pro #44: increase minimal rpm of mainboard fan inquiry

derFliegendeHamburger, 2024-06-18:
The original 40x40 mainboard fan had begun to make funny noises; so I replaced it.
Now when the printer firmware starts the fan for the first time, the voltage is a bit too low  - my fan starts but does not sound well. When later the firrmware decides to increase fan speed, all is well.
So my question is: where can I change/increase the minimal value for the running mainboard fan, please?
(just to be clear, this is not about the temperature trigger point, I am confident it was chosen well)

CChen616, 2024-06-19:
In the `printer.cfg` file, there's a section:
```
[controller_fan board_fan]
pin:U_1:PC4
max_power:1.0
shutdown_speed:1.0
cycle_time:0.01
fan_speed: 0.6
stepper:stepper_z,stepper_z1
```

When either the Z or Z1 axis stepper motors are on, the fan will start (hence, it starts at power-on since the Z axis is always locked). The fan will accelerate to max_power (up to 1) within a short time (kick_start_time defaults to 0.1 seconds) and then run at a power equal to fan_speed multiplied by max_power.

So, to increase the minimum speed, you should increase the fan_speed.

derFliegendeHamburger, 2024-06-19:
Your solution from above just does what I wanted. Thank you

closed 2024-06-19 by derFliegendeHamburger