QIDITECH/QIDI_Q1_Pro #46: Switch off all fans in standby/after cooling down at the end of printing

Deltaflyer7, 2024-06-20:
Is there a way to switch off all fans in standby mode / after printing? I still have fans running in standby mode, but I don't know if it's the motherboard fan or the power supply fan. But when I do a shutdown via the Fluid Console, all fans go off. I have the fans running in standby mode with both firmware V4.4.19 and V4.4.20.

CChen616, 2024-06-22:
Currently, there should still be two fans running in standby mode.

One is the board_fan. You can see in the `printer.cfg`:
```
[controller_fan board_fan]
pin:U_1:PC4
max_power:1.0
shutdown_speed:1.0
cycle_time:0.01
fan_speed: 0.6
stepper:stepper_z,stepper_z1
```
This indicates that the fan is associated with the Z-axis stepper motors. The fan will remain operational as long as the Z-axis is locked (the motors are always active to maintain the Z-axis height). In idle mode, the X and Y motors are unlocked. You can change the configuration to `stepper:stepper_x,stepper_y` to turn off this fan in idle mode. We may include this change into next release.

The power supply fan is controlled by the power module. It operates independently and cannot be changed currently.

Ameb, 2024-07-10:
> We may include this change into next release.

You did :). The issue can be closed.

Controlling the power supply fan would require a hardware revision of the printer.