QIDITECH/QIDI_Q1_Pro #49: Enable Beep on Display, diables speed button

MarkusKress, 2024-06-22:
The contrast on the pictures is not so good, but I think you can see it.

[image]
[image]

CChen616, 2024-06-22:
This icon may cause misunderstandings, but it is actually a silent mode switch. In this mode, the printing speed is reduced to minimize noise, making the speed non-adjustable. I will report this issue to the UI team.

CChen616, 2024-07-10:
The issue has been resolved in the 4.4.21 versions. If there are no other concerns, I will close this issue.

Ameb, 2024-07-10:
The setting is now located under the printing speed submenu. All it does is sending a `M220 S50` / `M220 S100` command?
I have created an idea discussion #64 to reduce noise from the steppers.