QIDITECH/QIDI_Q1_Pro #51: Mane or service not know

330Speedy, 2024-06-23:
[image]

konstantin-kelemen, 2024-06-23:
I did have the same issue, running `sudo dos2unix /etc/resolv.conf` fixed it.

330Speedy, 2024-06-23:
Great work!!! It works... Thank you very much :)
Mainsail is installed and up to date. But I can't open the web UI. The classic port 4408 doesn't work... Or does this install set another specific port?

pbyrne234, 2024-06-28:
Hi; any update on specific ports?

CChen616, 2024-06-28:
Checking which port is being listened to in `/etc/nginx/sites-available/mainsail` might be helpful. If it’s also listening on port 80, it may conflict with Fluidd.

330Speedy, 2024-06-28:
> `/etc/nginx/sites-available/mainsail`Může být užitečné zkontrolovat, na kterém portu se naslouchá . Pokud také naslouchá na portu 80, může to být v konfliktu s Fluidd.

Yes, I just checked and both mainsail and fluidd are on the same port 80. Please, how can I change the port? it didn't ask me during installation
Thank you

CChen616, 2024-06-28:
You only need to change the port used by one of the configurations to another, such as 4180 (you can use `netstat -ntlp` to check port usage and avoid conflicts), then restart the printer.

When accessing, use
```sh
# port can be omitted if service listening on port 80, which is http default port
http://{your_printer_ip}:{port}
# for example
http://192.168.0.2:4180
```

pbyrne234, 2024-06-28:
Hi can't find where KIAUH is installed? help appreciated.

pbyrne234, 2024-07-01:
@CChen616 Thanks, had already found it.  Fluidd needs to be updated as it never shows correct estimated finish time always only shows start time instead.

[image]

CChen616, 2024-07-02:
[image]

These Z-axis related buttons have been adjusted in the stock Fluidd interface. After your upgrade, their directions will be reversed:
The up arrow will make the platform move down.

330Speedy, 2024-07-07:
Hi everyone…. I have a little question 😉 
Is it possible to update klipper, moonraker, fluidd to the latest version? or if not, is there any plan to ship the latest version with the next FW update, or even native support for Mainsail? 
Thank you All

pbyrne234, 2024-07-07:
@330Speedy 
Mainsail is working fine when installed with KIAUH  when you change the port in sites-available.

😉

330Speedy, 2024-07-07:
> @330Speedy Mainsail is working fine when installed with KIAUH when you change the port in sites-available.
> 
> 😉

Yes, Mainsail works, but because of the old version of Moonraker and Klipper, not all options in Mainsail work...

pbyrne234, 2024-07-07:
@330Speedy Agreed they need to update both in next firmware.

CChen616, 2024-07-08:
The machine uses custom versions of Moonraker and Klipper. Upgrading will overwrite the modified files, causing the printer to malfunction. Please avoid upgrading these components.