QIDITECH/QIDI_Q1_Pro #52: 希望优化进退料流程，让进退料时喷头移动到废料盒，并增加擦料流程。

0XiaoSu0, 2024-06-26:
每次换料时废料总会落在打印面板上，希望在换料时喷头自动移动到废料盒位置，使得废料落入废料盒中，并在换料完成后执行一遍擦料过程，使废料落入废料盒并保持喷嘴干净。

CChen616, 2024-06-26:
由于Q1机型挤出机没有配备耗材切刀，难以将整个流程(退料 切换耗材等)完全自动化。

目前推荐的做法是手动移动喷头到指定位置，换料后手动擦料。

或者尝试将部分操作自动化，在`gcode_macro.cfg`中有：
```
[gcode_macro M603]
description: filament unload
gcode:
    M118 Heat up complete
    G92 E0
    G0  E15 F400
    G4  P1000
    G92 E0
    G1  E-90 F800
    M400
    M118 Unload finish

[gcode_macro M604]
gcode:
    M118 Heat up complete
    M83
    G1 E80 F400
    M400
    M118 Load finish
```

M603用于退料 M604用于进料 可以在宏起始位置加入
```
    G28
# 用于移动到废料盒
    G1 X97 F15000
    G1 Y243 F15000
    G1 Y254 F800
```
结束位置加入
```
# 需要指定温度
    CLEAR_NOZZLE HOTEND=220
```
示例:
```
[gcode_macro M604]
gcode:
    G28
    G1 X97 F15000
    G1 Y243 F15000
    G1 Y254 F800
    M118 Heat up complete
    M83
    G1 E80 F400
    M400
    M118 Load finish
    CLEAR_NOZZLE HOTEND=220
```
修改完后需要重启Klipper以应用

控制系统目前并没有适配，因此在屏幕上点击再次进料时，会重复归零擦料等，导致耗时增长

TinkerBarn, 2024-06-27:
> 由于Q1机型挤出机没有配备耗材切刀，难以将整个流程(退料 切换耗材等)完全自动化。
> 
> 目前推荐的做法是手动移动喷头到指定位置，换料后手动擦料。
> 
> 或者尝试将部分操作自动化，在`gcode_macro.cfg`中有：
> 
> ```
> 
> ```

Because you mentioned, that the Q1 has not a automated filament cutter, means that, if QIDI release a AMS (Multi Material Support) it will not be compatible with the Q1?

CChen616, 2024-06-27:
> Because you mentioned, that the Q1 has not a automated filament cutter, means that, if QIDI release a AMS (Multi Material Support) it will not be compatible with the Q1?

Yes

trivarialthea, 2024-06-27:
Wouldn't it be possible as an upgrade? There are already models that add this functionality manually.

See here for example https://www.printables.com/model/865926-remix-of-qidi-cutter-v20-body-for-q1-pro/comments

Though I can see it requiring a new board...

CChen616, 2024-06-27:
You can provide this suggestion to the support team. They will collect and record the relevant requests, and we will consider them in the future.

TinkerBarn, 2024-06-27:
Perhaps this can also be solved mechanically. Similar to the printable solution, an attachment on the extruder housing, which is screwed into the PTFE tube inlet and is then actuated by repeatedly moving the print head against a metal / plastic pin, which is located at the front, left or right in the printer housing (blade is pressed and cuts the filament).
The mechanical solution is similar to that used for cleaning the Nozzel.

This could be offered as a upgrade kit for Q1 Pro users and all X-Series printers.

But back to the topic.

Please cold you implement in a future firmware version for filament change in the display and in the gcode_macro.cfg, such a way that during a manual filament change without AMS, (which, by the way, is carried out in almost all slicers with M600 for HueForge prints, for example), the print head moves over the cleaning station.
The display ask the user to cut manually the filament and click on Next then the filament is unload/load. There is no unloading with retract. maybe only the option to select an unload (retract) manually in the display. But the standard should be only extrude. 
This means that the new filament would load with manual feeding by the user while the extruder extrude the filament.
The user can then also press repeat several times on the display until, for example, the new colour comes out cleanly.
Then cool and clean the print head. Head the nozzle to the original print temp and continue printing.

This would allow to control a standard pause with M0 in the Gcode and a filament change for a different colour with M600.

TinkerBarn, 2024-06-27:
> You can provide this suggestion to the support team. They will collect and record the relevant requests, and we will consider them in the future.

Thanks I have done it :-)

trivarialthea, 2024-06-27:
> > You can provide this suggestion to the support team. They will collect and record the relevant requests, and we will consider them in the future.
> 
> 
> 
> Thanks I have done it :-)

Keep me updated on their response, don't want to create multiple tickets for the same suggestion :)

TinkerBarn, 2024-06-27:
> Keep me updated on their response, don't want to create multiple tickets for the same suggestion :)
May you should it do, too. To give them the feeling, that is not only a wish from one customer...

TinkerBarn, 2024-06-28:
> > > You can provide this suggestion to the support team. They will collect and record the relevant requests, and we will consider them in the future.
> > 
> > 
> > Thanks I have done it :-)
> 
> Keep me updated on their response, don't want to create multiple tickets for the same suggestion :)

I get a response by mail:
_**Thank you very much for your suggestion, I've sent it back to the engineers.**_

CChen616, 2024-06-28:
This is just a template for the response. The main point is to convey to the decision-makers whether multi-color printing will be supported on the Q1 model.