QIDITECH/QIDI_Q1_Pro #53: Filament Sensor on QIDI Q1 Pro Continuously Reporting "Off" Despite Filament Present

hoangchieu91, 2024-07-01:
Today, my QIDI Q1 Pro is consistently reporting the filament sensor as "Off" even though there is filament present. I used the commands ENABLE_FILAMENT_WIDTH_LOG and QUERY_RAW_FILAMENT_WIDTH to check the sensor's returned values. The sensor is reporting values that significantly differ from the actual diameter of my filament. I am using 1.7mm PETG filament, and the sensor frequently returns values in the range of 0.6 to 1.12mm.

Steps to Reproduce:

Load 1.7mm PETG filament into the QIDI Q1 Pro.
Observe the filament sensor status, which shows "Off" despite the filament being present.
Use the command ENABLE_FILAMENT_WIDTH_LOG to start logging the filament width.
Use the command QUERY_RAW_FILAMENT_WIDTH to check the sensor's reported values.
Observe the reported filament width values.
Expected Behavior:
The filament sensor should correctly detect the presence of filament and report values close to the actual diameter of the filament, which is 1.7mm in this case.

Actual Behavior:
The filament sensor reports "Off" status even when filament is present, and the sensor returns values ranging from 0.6mm to 1.12mm, which is significantly off from the actual filament diameter.

Additional Information:

Printer Model: QIDI Q1 Pro
Filament Type: PETG
Filament Diameter: 1.7mm
Any assistance to resolve this issue would be greatly appreciated. Thank you!
[image]
[image]

CChen616, 2024-07-02:
Here, the on/off indicator shows the filament sensor's activation status (it automatically turns off when a pause is triggered). If you have enabled the park mode in the timelapse settings, it can cause the print to repeatedly pause and resume, causing the sensor to continuously turn on and off.

Regarding the filament diameter, the detected value is only used to check if filament is present; as long as it is greater than 0.5mm, it will not trigger a filament runout. The large variation in readings is due to insufficient space above, causing the filament to enter the extruder at a non-vertical angle, which affects the measurements. However, as long as the minimum value does not fall below 0.5mm, false triggers will not occur.

hoangchieu91, 2024-07-02:
I did not set a pause for the time lapse.
[https://youtu.be/hYoM5MNhN80](url)
 That day, my print automatically paused multiple times. It's possible that the sensor dropped below 0.5 at times, but I couldn't catch it.

Additionally, in your response: If the filament does not go straight but at an angle, the measured diameter will be larger than 1.75 (the actual diameter of the filament, which I measured with a caliper), It should not be within the range of 0.6-1.2 and often around 0.8-1.0 like this. It feels like the parameter is scaled by 0.5 times.
This is the logging:

> 10:44:28 $ QUERY_RAW_FILAMENT_WIDTH
> 10:44:28 // ADC1= 5002.0 ADC2=8456.0 RAW=13458.0
> 10:44:29 // Filament width:0.846
> 10:44:33 // Filament width:0.818

This is the config on my printer:

> [hall_filament_width_sensor]
> adc1: gpio27
> adc2: gpio28
> cal_dia1: 1.50
> cal_dia2: 2.0
> raw_dia1: 14397
> raw_dia2: 15058
> default_nominal_filament_diameter: 1.75
> max_difference: 0
> measurement_delay: 50
> enable: false
> measurement_interval: 10
> logging: False
> min_diameter: 0.3
> use_current_dia_while_delay: False
> pause_on_runout:True
> runout_gcode:
>             RESET_FILAMENT_WIDTH_SENSOR
>             M118 Filament run out
> event_delay: 3.0
> pause_delay: 0.5

I think you need to adjust the parameters to calibrate the measurement values more accurately."

Alhamzah93, 2024-08-23:
Am facing same problem the runout sensor get off during printer(in mid print)
Try to change the min_diameter to 0.1 but still facing same problem
Try to disassemble the sensor clean it with on use

CChen616, 2024-08-23:
Is the filament width sensor giving false alerts during printing? (Meaning there is filament, but the sensor thinks there isn't and pauses the print.) If so, did this issue start recently, or has it been happening from the beginning?

Try entering `ENABLE_FILAMENT_WIDTH_LOG` before starting the print, and check what output you see in the Fluidd console.

Alhamzah93, 2024-08-23:
15:49:11 
// Filament width:0.496
15:49:22  
// Filament width:0.495
15:50:00 
// Filament width:0.497
15:50:29 
// Filament width:0.490
15:50:54 
// Filament width:0.552
15:51:08 
// Filament width sensor Turned Off
15:51:12 
// idle_timeout: Timeout set to 86400.00 s
15:51:13 
// Filament width measurements cleared!
15:51:13 
echo: Filament run out
15:51:24 
// idle_timeout: Timeout set to 43200.00 s
15:51:43 
// Filament width sensor Turned On
15:51:43 
// Filament width measurements cleared!
15:51:43 
// Filament NOT present
15:51:48 
// Filament width sensor Turned Off
15:51:51 
// idle_timeout: Timeout set to 86400.00 s

​

CChen616, 2024-08-24:
It might be a hardware issue. Open the front cover of the extruder and check if the spring in the area shown in the picture is loose, if there's any blockage in the middle, or if the wiring of the Hall sensor is loose. 

If the issue persists, contact customer support for a part replacement.

[image]

Alhamzah93, 2024-08-24:
Already checked that it's loose...now made the min width 0.0 it's working but I think if the filament run out it will not stopped.