QIDITECH/QIDI_Q1_Pro #55: Filament Load Routine on printer panel stalls after loading with 4.4.21

nobrainer32, 2024-07-02:
After online update my Q1 Pro from 4.4.20 to 4.4.21 i have startet to load new filament. 
I used the same routine on the printer panel as before but now it hangs after successfull load at step 3. 
Only reboot enables the panel response again. Unfortunately, this behavior is reproducible even after a restart.

[image]

CChen616, 2024-07-03:
Thanks for pointing out the error.

You can fix this by modifying the `gcode_macro.cfg` file. Change `[gcode_macro M603]` and `[gcode_macro M604]` from:

```
    M118 Filament unloaded
......
    M118 Filament loaded
```
to:
```
    M118 Unload finish
......
    M118 Load finish
```
Then, restart the printer to apply the fix.

nobrainer32, 2024-07-03:
Thank you - just tried the fix and it is working fine :-)

dewi-ny-je, 2024-07-03:
> Thanks for pointing out the issue.
> 
> You can fix this by modifying the `gcode_macro.cfg` file. Change `[gcode_macro M603]` and `[gcode_macro M604]` from:
> 
> ```
>     M118 Filament unloaded
> ......
>     M118 Filament loaded
> ```
> 
> to:
> 
> ```
>     M118 Unload finish
> ......
>     M118 Load finish
> ```
> 
> Then, restart the printer to apply the fix.

I downloaded the source of 4.4.21 and I can still see the old, wrong text, even if the Release page makes me think the manual modification is needed only for people who updated "already".

Maybe releasing 4.4.22 with everything corrected would be easier.

CChen616, 2024-07-03:
The configuration files in the `config` folder in the repository are for record-keeping only and are not the source files used during packaging. They simply haven't been synced yet.

The files within the update package have already been corrected.

dewi-ny-je, 2024-07-11:
Then this ticket should be closed

closed 2024-07-11 by nobrainer32