QIDITECH/QIDI_Q1_Pro #56: Pre-print sequence doesn't equal up Z-heights of each Z motor rod anymore

Pheggas, 2024-07-03:
Hello. Title might sound a bit strange but when the printer came on v4.4.16 firmware, it has some kind of measurement for height difference between each Z-axis rod. It went probed height of right rod first and then moved to left rod, made another measurement and then it equaled up those differences so the buildplate is perfectly straight between left and right rod. It doesn't do this procedure anymore in v4.4.19.

CChen616, 2024-07-05:
Even if the old leveling method ensures that the PEI plate is level, the adjustment screws might cause the platform to be at an angle with the axis. If this angle is too large, it could hinder the platform's movement.

[image]

In the new version, by clicking `Platform Calibration` on the calibration page, the platform will first move downwards to zero position and then proceed with screw adjustments. This aims to ensure both the PEI plate is level and the platform is perpendicular to the axis.

Pheggas, 2024-07-05:
That makes sense. I re-ran the calibration and it was horrible to see the difference between sides were like 1 cm. Thank you so much for detailed explanation even with image for better understanding. Now i know i should run platform calibration occasionally or after some disaster like power outage. Closing this issue.

closed 2024-07-05 by Pheggas