QIDITECH/QIDI_Q1_Pro #57: Wfi completely lost

kerwalker, 2024-07-03:
Since updating to 4.4.21 I have lost wifi server link to the machine. Rebooted computer, printer and network router and zero. Also tried to re configure the network settings on printer. To say it was sluggish is an understatement. Lag was unbearable and was freezing  up the printer so had to unplug from mains power but still can't get the server to upload files. total garbage update!
[image]
The file shows the error message I receive after an attempt to upload a stl file. And just for interest, Fluidd and Moonreker are working with the Qidi Link and my browser with no issues.

Lets hope there is a solution to this issue!

rav101, 2024-07-03:
Just to check if you have any connection to the printer? I found once I updated to .21 my IP for the router changed so just sense checking that it's not available but on a different address?

kerwalker, 2024-07-03:
I checked router sees the device and I can connect to it over the network and it all works as it should on Fluidd and Moonraker.

CChen616, 2024-07-05:
Multiple attempts have failed to reproduce the issue of IP address reassignment after an update. Providing any of the following information would help us identify the problem:

1. Was the update done online or offline?
   Offline updates require a printer restart, which resets the Wi-Fi connection.
   
2. Are the router settings such that DHCP is enabled?
   When enabled, the router dynamically assigns IP addresses but generally prefers to reassign the same IP to the same device and ensures the IP remains unchanged during the lease period.
   
3. Have you configured DHCP static IP assignments?
   If a static IP is assigned to the printer, that IP address will be reserved.
   
4. Is the number of devices connected to the router exceeding the range of DHCP-assignable IP addresses?
   If the limit is exceeded, the IP address might be reassigned to other devices when the printer disconnects.

xsw21qay, 2024-07-05:
Same here... In order to fix the issue, I had to connect to a different network and then back to the original one. Now it works again.

rav101, 2024-07-05:
> Multiple attempts have failed to reproduce the issue of IP address reassignment after an update. Providing any of the following information would help us identify the problem:
>
>     1. Was the update [...]

If that one was for me, I wouldn't worry about it. There are so many factors here with home router setups. I've now assigned static IP and it's not a big issue.

xsw21qay, 2024-07-05:
Bizarre enough: The printer was able to retrieve its IP-address through DHCP (I configured a static lease so it's always getting the same address) but other than that packages sent to the printer (even simple ICMP packages like ping) wouldn't go through.

I had to switch to another network first and then back again to get it working again (on a sidenote: Please consider a disconnect, discard or ignore option for the selected wifi network, that's easier than switching to another network first :) ).

CChen616, 2024-07-05:
Starting from version 4.4.18, updates only restart related services (klipper, moonraker, screen_control_system) and do not prompt for a printer reboot to complete the update. Generally, the network status should remain unchanged. If the IP address changes after a restart, it could indeed be related to different router settings. Currently, I can't reproduce the issue neither think of a better solution, but I will add the relevant instructions to the repository documentation.

Adding a `Forget Wi-Fi Connection` button is a valuable suggestion, and we will consider it.

kerwalker, 2024-07-05:
Thanks for the info provided. I’m not at home until tomorrow so will do a re assignment of the ip range and hopefully this sort out the issue. It is strange it works on fluidd and connects to it no issue but when sending a file it pulls a server error as per the screenshot. I found that very odd. Sent from my iPhone

xsw21qay, 2024-07-05:
> Starting from version 4.4.18, updates only restart related services (klipper, moonraker, screen_control_system) and do not prompt for a printer reboot to complete the update. Generally, the network [...]

Oh, this didn't happen right after the update. After the update I was able to print a couple of things and this morning when I turned the printer on all of a sudden the printer couldn't receive any data anymore.

Raxasus, 2024-07-05:
I believe I am having a simmilar issue, my printer stopped mid print and once it does it just keeps looping trying to reconnect, it happened to me 3 times now.. I managed to print a few things fine after the .21 update but not much as the time when I updated it seemed simmilar I do believe it is somehow related to the .21 update.
[image]
[image]
[image]

kerwalker, 2024-07-05:
Mine prints ok with no issues via usb but was not over the network. Try printing direct via usb to confirm it’s not or is the network issue that’s causing the shutdown. Sent from my iPhone

3.PNG (view on web)
2.PNG (view on web)
1.PNG (view on web)

kerwalker, 2024-07-05:
have resolved my issue of no wireless network. Although it does not solve the issue on Qidi Slicer, I do have my printer up and running on the wireless network using Orca Slicer. It is a little more involved in setting it up but there are Qidi profiles for the Q1 pro and it working as normal. But the question is, why can this happen? My guess is the file is written to a memory location that is possibly corrupt or unusable somehow and Orca Slicer writes to a different location in memory. It works on both my MacBook and Windows machine. It's bizarre this is the case. Anyway thanks to all who contributed with info to try and resolve the issue.