QIDITECH/QIDI_Q1_Pro #59: Printer stops mid print since last update

Raxasus, 2024-07-05:
My printer started to stop mid print saying something like klippy has been restarted, trying to open up the web interface hardly connects and it also has issues with wifi conectivity and transfer rate compared to before the .21 update. It keeps looping on connecting to moonraker and it also from time to time connects for just a moment before it shows the message "Klippy not connected."
[image]
[image]
[image]

CChen616, 2024-07-05:
It looks like the system rebooted due to an overload. Did the board fan run properly during printing?

If you can SSH into the printer system, run the `uptime` command. When was the last system startup?

Raxasus, 2024-07-05:
I am not at home at the moment, I did leave it on overnight and it did a print fully, before that print 2 tests failed as it stopped and this morning I had it start another, it really was not that hot compared to the days before here but I believe the fan was working properly through I have no way of checking, could you provide more info on how to connect through ssh? I tried a ping but I got loss packets to it..
[image]

CChen616, 2024-07-05:
If you cannot ping the printer, it might be due to an IP address change or a Wi-Fi disconnection. You can log in to your router's management interface to check the connection status of the device named `mkspi`.

Once you have determined the IP address, use an SSH tool (recommend MobaXterm) to start a new session. Specify the remote host as the printer's IP address and log in with the username `mks` and password `makerbase`.

After running the command, you will see an output similar to this:
```sh
mks@mkspi:~$ uptime
 18:10:21 up  3:54,  1 user,  load average: 1.68, 1.57, 1.55
```

Here, 3:54 means the system has been running for 3 hours and 54 minutes.

Raxasus, 2024-07-05:
The current uptime wont do much for me then since after it happened I first tried to "restart" it by cutting the power from the wall socket and it still did not work, it was somewhere around 12-15 hours through I assume since it had to do a print for 5 hours total and as I was sleeping it did finish that print and sit idle for 4-5 hours or so before I started this one that failed mid print.

CChen616, 2024-07-05:
Printing begins after the file transfer is complete, meaning the entire content of the gcode file should already be present locally when printing starts. 

The components such as Klipper, Moonraker, and Fluidd communicate via the local loopback address 127.0.0.1. Issues with other network connections should not affect the operation of these services. Therefore, if the printing process terminates abruptly and there are no relevant logs in the `klippy.log`, it is likely that the system encountered an issue, such as a shutdown due to overload.

If possible, please attach the relevant log files. You can export the logs using the `Exporting Logs` button in `system message` page on the screen.

Raxasus, 2024-07-05:
I will try to attach them when I get home then and fully restart the printer manually as it just loops for me and I cannot download it in the brief moment it does access the web interface.

Raxasus, 2024-07-05:
After several tries I managed to get the moonraker log downloaded if that helps at all, I will reply with the klippy logs when I manage to get them.
[moonraker.log](https://github.com/user-attachments/files/16108681/moonraker.log)

Raxasus, 2024-07-05:
Managed to get the klippy log but it is to big to post here

Raxasus, 2024-07-05:
https://tmpfiles.org/dl/8839986/klippy.log

Raxasus, 2024-07-05:
Also the time from the klipper interface seems to be wrong for some reason, it managed to reconnect just now, in the settings the time is correct but in the logs it is not or even in estimating when the print would be done.

An example from the config file of the time:

Args: ['/home/mks/klipper/klippy/klippy.py', '/home/mks/klipper_config/printer.cfg', '-I', '/tmp/printer', '-l', '/home/mks/klipper_logs/klippy.log', '-a', '/tmp/klippy_uds']
Git version: 'v0.10.0-530-g3387a9c2-dirty'
CPU: 4 core ?
Python: '2.7.16 (default, Oct 10 2019, 22:02:15) \n[GCC 8.3.0]'
Start printer at Fri Jul  5 17:17:19 2024 (1720171039.9 25.5)
===== Config file =====

[image]

Raxasus, 2024-07-05:
The config file is some hours ahead compared to the fluidd settings time.

CChen616, 2024-07-05:
Fluidd is likely using the time from the device running your browser, while the time in the logs is based on the printer system. You can configure the timezone to match by using the command `sudo dpkg-reconfigure tzdata`.

Raxasus, 2024-07-05:
Can I run the above command from the fluid interface? 
[image]

CChen616, 2024-07-05:
That won't work well because the command has an interactive step to select the timezone. You can check the `time_update.cfg` file in the Fluidd configuration page, which contains instructions on how to set the timezone through Fluidd.

kerwalker, 2024-07-05:
Now got further issues. The USB port no longer accessible on the machine. I have switch it on in the settings but does not read any files

CChen616, 2024-07-05:
It is possible that frequent and rapid plugging and unplugging of the USB drive has caused the drive to become corrupted. Please try formatting the USB drive on a PC. Any file system format will do.

kerwalker, 2024-07-05:
I have tried that and it’s still the same. I’m going to open it up and change it over to the spare usb on the panel to see if this is the issue. The strangest thing is in fluidd if I take a file I have already printed I can send the file and it prints. This now eliminates the router or network issue. Very strange. Sent from my iPhone

Raxasus, 2024-07-05:
Do the klipper logs above that I provide point towards whatever might have been the issue for my failed prints? I did look through them a bit but by the way them ended I assume it just.. stopped without any error message and such?

Also quick question, I used some filament for a print and it went out at 90% of the print or so, after changing it and pressing resume the new filament used just did not properly stick to the other one and I had printed so far so it mostly failed, any way I can improve that?

Raxasus, 2024-07-05:
Also I just tried to send another print file to the printer and it just stoped at the point where it should clean the nozzle and did nothing,I could see it trying to connect on the browser but it did not so I refreshed it and it just did not start at all , here is the klipper log:
[klippy_latest.log](https://github.com/user-attachments/files/16112021/klippy_latest.log)

Raxasus, 2024-07-05:
Status is saying in progress but it is doing nothing

[image]

kerwalker, 2024-07-05:
I have resolved my issue of no wireless network. Although it does not solve the issue on Qidi Slicer, I do have my printer up and running on the wireless network using Orca Slicer. It is a little more involved in setting it up but there iare Qidi profiles for the Q1 pro and it working as normal. But the question is, why cn this happen?  My guess is the file is written to a memory location that is possibly  corrupt or unusable somehow and Orca Slicer writes to a different location in memory. It works on both my MacBook and Windows machine. It's bizarre this is the case. Anyway thanks to all who contributed with info to try and resolve the issue.

CChen616, 2024-07-06:
[image]
[image]
This is what I see when I turn off the power during printing and then turn it back on. It's basically the same as what you uploaded.

It's difficult to pinpoint the cause at the software level. I suspect it may be related to hardware or the usage environment, such as unstable voltage, high temperatures, loose power connections, etc. If, after ruling out these factors, the printer still stops unexpectedly, please contact customer support for a motherboard or other hardware replacement.

Raxasus, 2024-07-06:
Yeah.. it does indeed feel like it just turns off by itself or something when it does that, it did ask me on its screen if I want to resume printing but the electricity itself is quite stable and had no blackouts while it happened, I will do that and speak with the customer support, it just felt strange and the timing was quite close to the latest update truth be told.

As I was checking all the fans and such I actually found that if you turn on the chamber heating fan by setting a temperature it starts spinning but if you then turn it back to 0 the fan itself wont stop, is that normal? Also the motherboard fan for my printer turns on only if the extruder one is turned on as well otherwise it is idle, maybe more control for those could be a nice feature for the future.

kerwalker, 2024-07-06:
The sensor for monitoring tangled filament was kicking in for me a few weeks ago and the only thing that came up on the screen was a resume. In my case it was a tight filament reel, but resume does not tell you much. I just checked the reel and it was indeed tight but it was eliminating one thing at a time to see if I could find the issue.

kerwalker, 2024-07-06:
Also I also added a comment that the USB failed. This was a result of the 8GB supplied USB failing. Change to a branded USB and all is fine now.

Raxasus, 2024-07-08:
> That won't work well because the command has an interactive step to select the timezone. You can check the `time_update.cfg` file in the Fluidd configuration page, which contains instructions on how to set the timezone through Fluidd.

I did the config and it did work well for a while but after I restarted and then started the printer once again the times got reset I suppose, I am not sure if the motherboard of the printer has any battery and such to be able to save the correct date/time while the printer is offline, maybe as a future feature there could be a setup for this as in selecting the timezone and the printer once it is connected to the wifi automatically updates its date/time?

CChen616, 2024-07-08:
The script includes steps to enable the NTP service, allowing the system to automatically sync time via the network.

Adding timezone configuration during onboarding guide is a possible option, but the implementation details are still under consideration. The challenge lies in the numerous available timezones, making it difficult to present them in the UI for selection.

Raxasus, 2024-07-08:
> The script includes steps to enable the NTP service, allowing the system to automatically sync time via the network.
>
> Adding timezone configuration during onboarding guide is a possible option, but [...]

# RUN_SHELL_COMMAND CMD=UPDATE_SYSTEM_TIME PARAMS={TimeZone}

Is the only thing the time_update.cfg provides except the list of available timezones and an example, the menu could be made from klipper only maybe or if UI is input is required.

Also maybe for the ui a simple idea would be to add a small menu to set the GMT value in numbers not a list with all of the timezones and their names.