QIDITECH/QIDI_Q1_Pro #60: Printhead should home first for filament replacement

wrobelda, 2024-07-07:
As in the title: the manual filament replacement should move the printhead to the center to be able to cut it off. Right now I have to do this manually before the replacement, otherwise I end up with this:
[image]

Raxasus, 2024-07-07:
Center but also lower the print bed some more to have more access to take the filament out from beneath the nozzle maybe?

wrobelda, 2024-07-07:
Yes, good point

CChen616, 2024-07-08:
You can home first before replacing the filament, or you can unlock the motor and manually move the extruder to the center or above the waste bin.

Generally, when a printer is idle and not homed, the motors are unlocked, allowing you to manually move the extruder. If you include movement in the process, the printer will need to home first, which takes more time. In most cases, manually moving the extruder by hand is a better option.

wrobelda, 2024-07-09:
> Generally, when a printer is idle and not homed, the motors are unlocked, allowing you to manually move the extruder. If you include movement in the process, the printer will need to home first, which takes more time. In most cases, manually moving the extruder by hand is a better option.

That's fair, but you did add a convenient UI option to replace the filament which inevitably must confuse many users as to whether the printhead's position is in the right place or not. Surely many of them will not think they are expected to move the head themselves at this point, especially to move it manually.