QIDITECH/QIDI_Q1_Pro #62: Consider adding mobileraker companion, or at least bundling the necessary dependencies for it by default

wrobelda, 2024-07-09:
I can see you use kiauh for building Klipper and other components, so I would suggest that you also bundle the mobileraker companion by default. I am aware you have your own mobile app, but some definitely prefer to use mobileraker, so I suppose bundling and adding it commented out in moonraker.conf would do the job?

CChen616, 2024-07-10:
Due to Mobileraker's license prohibiting commercial use, we have decided not to install the software by default to avoid any risks. 

We recommend that users who need it choose and configure it themselves to ensure compliance with their own needs and usage terms.

wrobelda, 2024-07-10:
I'm not sure I follow, the mobileraker-companion is licensed with a permissive MIT, which allows for unrestricted commercial use?

For the record, the companion software allows Mobileraker users to receive push notifications on their phones.

CChen616, 2024-07-10:
I mistakenly thought https://github.com/Clon1998/mobileraker was the plugin you recommended. Mobileraker_companion, serving as the client, does allow commercial use.

Considering the following points, we might still decide not to include this plugin in the default bundle:

- (The main reason) We already have our own mobile app support. Unlike plugins such as moonraker-timelapse, not including it won’t result in a loss of functionality, so it isn’t essential.
- Including too many such plugins by default can bloat the system. Additionally, we need to consider the ongoing maintenance issues (handling plugin problems and whether to keep up with their updates).
- Installation can be easily done through kiauh. Currently, our system is open for user customization, so if there’s a need for it, users can install it themselves.