QIDITECH/QIDI_Q1_Pro #65: Improve Hall sensor calibration

dewi-ny-je, 2024-07-12:
I calibrated at room temperature my filament sensor using drill bits 1.5, 1.6, 1.7, 1.8, 1.9, 2.0 mm (verified with a micrometer) and I found that the sensor behaviour is not linear, and the values provided by Qidi are not correct. 
It seems that QIDI calibrated using 1.50 and 2.0 mm and the values provided by QIDI match mine reasonably, the issue is that the filament sensor gives almost flat readings above 1.85 mm, and QIDI didn't notice, so the curve Klipper would calculate is not accurate.

See the curve I measured, my proposal and your original values:

[image]

So basically my suggestion is to change the settings in https://github.com/QIDITECH/QIDI_Q1_Pro/blob/a24c5dbe72544077adeb7aeee394e6b8c4d16eca/config/printer.cfg#L71

to 

```
cal_dia1: 1.50
cal_dia2: 1.80
raw_dia1: 13868
raw_dia2: 14918

```

You will notice that the sensor is useless above 1.80 mm, but it doesn't matter: filaments are basically never that thick, see experiment from https://github.com/Deutherius/Filament-Width-Comp-Experiments (graph at the end, at most 1.78 mm).

It might not be perfect and I agree it should be off by default, but it's a clear improvement over the current settings for those who would like to play with it.

CChen616, 2024-07-15:
Thank you for pointing out the issue and providing a solution. We will proceed with further testing to determine the appropriate calibration values.

dewi-ny-je, 2024-07-15:
> Thank you for pointing out the issue and providing a solution. We will proceed with further testing to determine the appropriate calibration values.

Unfortunately I noticed that the filament path before the sensor is not constrained enough to ensure the filament is properly straight and this means the readings are not reliable once the printer is on operation

CChen616, 2024-07-15:
Yes, due to the mechanical structure, the values read by the Hall sensors will vary between different machines. As observed previously, further calibration can only ensure that the boundary values are kept away from the threshold that triggers the filament runout detection, but it cannot accurately reflect the actual filament diameter.

closed 2024-08-08 by dewi-ny-je