QIDITECH/QIDI_Q1_Pro #69: Dynamic adjustment of pressure advance

dewi-ny-je, 2024-07-18:
Klipper recently implemented a correction which allows changing pressure advance on the fly without blobs or print defects:
https://github.com/Klipper3d/klipper/pull/6635

Thanks to it, a feature of Orca can now be used:
https://github.com/SoftFever/OrcaSlicer/pull/5609

Let me show you some examples of what the feature allows:

WITH DYNAMIC PA:

Top surface

[image]

Solid infill

[image]

WITHOUT DYNAMIC PA (pressure advance calibrated, as usual, for corners only):

[image]

The improvements are huge and I think it's worth to backport the Klipper commit linked above.

CChen616, 2024-07-19:
We will continue to monitor this feature, and if it proves to be stable and without issues, it will be included in a future update.

dewi-ny-je, 2024-07-19:
> We will continue to monitor this feature, and if it proves to be stable and without issues, it will be included in a future update.

I plan to apply the patch manually in the next days. I'll report back after performing at least 10 prints with it enabled, maybe this will help you evaluating it in real world scenario.

dewi-ny-je, 2024-07-31:
Right, I forgot it also requires flashing.
So basically this new feature will never come to existing Q1 printers? that's a pity.

I'll do as suggested.

Am I right that, once new firmwares for the Q1 are released by QIDI, I'll just need to manually patch extruder.py without recompiling/flashing the main board and the extruder board?

CChen616, 2024-07-31:
> Am I right that, once new firmwares for the Q1 are released by QIDI, I'll just need to manually patch extruder.py without recompiling/flashing the main board and the extruder board?

Yes, and to be safe, it's best to back up the `extruder.py` file before updating, as we might modify this file in the new patch.

dewi-ny-je, 2024-11-24:
@CChen616 
> For extruder board: 
> Press and hold the `boot` button on the extruder board and start the machine. The extruder board should mount as a USB device, and the device directory should be mounted to `gcode_files/sda1/` (assuming there are no other USB devices). Move the generated `klipper.uf2` file into this directory and then reboot.

To be clear: the Q1 extruder board seems connected via serial, not USB, see printer.cfg:

```
[mcu U_1]
serial: /dev/ttyS0
restart_method: command
```

Will it really appear as USB mass storage, since it's apparently not USB to begin with?

I was planning to do it now but I was pointed out to this inconsistency.

I later found also https://github.com/frap129/OpenQ1/blob/main/docs/Installation.md which instructs (step 3) to solder new wires to the toolhead for the purpose.

Thanks for the help

dewi-ny-je, 2024-11-29:
I was right, see https://github.com/QIDITECH/QIDI_Q1_Pro/issues/78#issuecomment-2307991617

dewi-ny-je, 2024-12-25:
I patched the firmware of my Q1 v4.4.24 and it works.
As you can see from my previous post and the wiki page the results are very good and I think worth a firmware release because the print quality is quite impressive.

https://github.com/SoftFever/OrcaSlicer/wiki/adaptive-pressure-advance

This is the patch without the compiled .pyc files, only the sources I modified.
[dynamic-PA.zip](https://github.com/user-attachments/files/18247077/dynamic-PA.zip)