QIDITECH/QIDI_Q1_Pro #73: chamber exhaust fan in version 4.4.21

Pukkahq, 2024-08-04:
I have replaced the exhaust fan with a carbon filter,  In the latest version the config has changed so the fan is running even when printing PLA. What's the best way to disable the fan as in orca slice I can only set the fan speed to 0 and not off.

**updated configuration file

Enable chamber exhaust fan by default when chamber temp set to 0. Board fan associated with x y motor drive instead of z axis.**

CChen616, 2024-08-05:
Try inserting `M106 P3 S0` into the start gcode or layer change gcode in Orca Slicer.

I'm not quite sure what you mean by replacing the exhaust fan with a carbon filter. Does the carbon filter come with its own fan? If that's the case, I'm not certain if the above solution will work.

Pukkahq, 2024-08-05:
I will give that a go, just a question if we are printing ASA will that setting overwrite the one set for the material?

This is the filter I am using

https://www.printables.com/de/model/909544-qidi-q1-pro-mount-for-the-nevermore-v6

CChen616, 2024-08-05:
In Orca Slicer, filament selection is not part of the preset. If you use the same printer preset, the start G-code and other settings won't change regardless of the filament selected. 

If you want to disable this feature (which keeps the exhaust fan on even without setting chamber temperature to reduce the risk of nozzle clogging in some situations), you can edit `gcode_macro.cfg` as follows:

```cfg
[gcode_macro PRINT_START]
gcode:
    AUTOTUNE_SHAPERS

    {% set bedtemp = params.BED|int %}
    {% set hotendtemp = params.HOTEND|int %}
    {% set chambertemp = params.CHAMBER|default(0)|int %}
    set_zoffset

    M104 S0
    M140 S{bedtemp}
#    M141 S{chambertemp}
    G28    
    CLEAR_NOZZLE HOTEND={hotendtemp}

    M190 S{bedtemp}
#    Z_TILT_ADJUST
    G29
    G0 Z50 F600
    G0 X0 Y0 F6000

    M109 S{hotendtemp}
    M141 S{chambertemp}
    M204 S10000
    {% if chambertemp == 0 %}
        M106 P3 S255
    {% endif %}
    ENABLE_ALL_SENSOR
    save_last_file
```

Delete:
```cfg
    {% if chambertemp == 0 %}
        M106 P3 S255
    {% endif %}
```
Then save and restart klipper.

dewi-ny-je, 2024-08-08:
> I have replaced the exhaust fan with a carbon filter, In the latest version the config has changed so the fan is running even when printing PLA. What's the best way to disable the fan as in orca [...]

The exhaust fan is surely meant to be activo with PLA, since it takes away warm air to improve prints and reduce likelihood of clogs.

It's off for high temp materials...