QIDITECH/QIDI_Q1_Pro #74: Compatibility third party WiFi dongles

dewi-ny-je, 2024-08-06:
Hello,
I need to connect the printer to a 5 GHz network and the dongle provided doesn't support 5 GHz. Which dongles can I use? 

If you don't have a list of known working devices, can you provide some generic help?

For example, would every dongle already supported on the raspberry pi (so, with drivers for ARM linux) work also on the Q1 without needing to compile anything?
Or, is there a folder in the MKS board with the list of supported chipsets so I can decide based on chipsets instead of product names?

Also: would I be able to use the display to set the wifi, or would I need SSH if I change dongle?

Thank you

CChen616, 2024-08-07:
The hardware engineer said that the Q1 board does not support 5GHz wireless networks.

The system uses `wpa_supplicant` and `wpa_cli` to manage and connect to wifi networks. Changing the network driver might result in a change of network interface names, which could cause issues because the system relies on fixed names like `wlan0` and `eth0`

For 5GHz support, you might consider dongles such as:

TP-Link Archer T2U Nano
Edimax EW-7822ULC
Netgear A6100
However, it's not certain whether these adapters are compatible with the Q1 board.

You might find more useful information by reaching out to the community to see if others have successfully made similar changes.

dewi-ny-je, 2024-08-07:
So three out of those four use the chip RTL8812BU, and one RTL8822BU.
I guess that as long as the chipset is the same, no issues will arise. I'll do a test.

dewi-ny-je, 2024-08-07:
TP-link T3U (commercial name: "AC1300 mini wireless MU-MIMO USB adapter") works.

closed 2024-08-08 by dewi-ny-je