QIDITECH/QIDI_Q1_Pro #75: Free 2-pin red port on the QIDI Q1 Pro adapter plate

KosVas91, 2024-08-13:
Hello! I have a question about a free 2-pin red port on the QIDI Q1 Pro adapter plate (small board in the printers head).
According to [this manual](https://wiki.qidi3d.com/en/Q1/Manual/Extruder-fan-installation) this port is designed for an optional extruder fan, and it 5V although all the rest fan ports are 24V. 

I would like to confirm that this port really is 5V one. Also does this port corresponds to this part of the printer.cfg?
[heater_fan hotend_fan2]
pin:gpio11

If it so, it is a great option to replace loud stock hotend cooling fan with a more powerful and silent 5V Sunon fan (something like MF25100V1-1000U-A99).

CChen616, 2024-08-14:
The corresponding output port according to this board diagram is 5V IO11.

[A-7 V1.0_002 PIN.pdf](https://github.com/user-attachments/files/16606570/A-7.V1.0_002.PIN.pdf)

KosVas91, 2024-08-14:
Thank you. This diagram is the best answer you could provide!

closed 2024-08-14 by KosVas91