QIDITECH/QIDI_Q1_Pro #76: Can you add features to monitor/report total power consumed during a print?

Rawrkanos, 2024-08-21:
I would like to be able to see how much power is used by a print. That's all.

CChen616, 2024-08-22:
The current hardware configuration does not support this feature.

You may consider using a smart plug with energy monitoring features, or alternatively, you can try to shut down other devices and check the power meter to estimate energy consumption.

hoangchieu91, 2024-08-23:
If you're interested, I have some data to share:

29W in standby mode.
350W when starting up (heating the bed and cleaning the print head, which takes about 60 seconds).
160W-190W during the entire printing process.
If you leave the printer plugged in without printing for a day, it will consume about 0.7kWh.

Can you calculate that?
[image]