QIDITECH/QIDI_Q1_Pro #77: Issue: Z-Offset Error

hoangchieu91, 2024-08-23:
Printer: Qidi Q1 Pro
Firmware: 4.4.21

Problem: The print head is either too close to or too far from the bed.

Details:
When I start a print, the printer automatically cleans the nozzle and calculates the Z-offset, performing an auto bed leveling. However, during printing, the print head sometimes ends up too far from the bed or too close, forcing me to manually adjust the Z-offset (ranging from -1.2 to 1.0). I've tried setting the Z-offset, but every time I start a new print, the printer resets it and recalculates the Z-offset (with incorrect results).

Consequence:
If I'm not quick enough to manually adjust the Z-offset, the print head either scrapes the print bed or ends up too far from the bed, causing the filament not to stick to the bed and instead clump around the nozzle.

CChen616, 2024-08-23:
In some cases, the printer's automatic leveling might have errors (such as interference during the leveling process). However, if this happens frequently, it could be due to excessive wear on the small wiper block used for nozzle cleaning. This wear can result in the nozzle not being properly cleaned after preheating, leading to inaccuracies during the leveling process.

You can check if there is any residual filament on the nozzle after the automatic leveling completes. If you find any, it's recommended to replace the wiper block.

hoangchieu91, 2024-08-23:
"Your explanation doesn't make sense when the printer crashes the print head into the bed. Can you show me where in the G-code the Z-offset value is taken? I want the printer to print out the value it has calculated in the terminal. After auto bed leveling, even though the bed mesh state is saved to the profile, the Z-offset value remains 0.

Example log:
19:47:05
// Bed Mesh state has been saved to profile [kamp]
// for the current session. The SAVE_CONFIG command will
// update the printer config file and restart the printer.
19:47:05
// Variable Saved
19:47:05
// Bed Mesh state has been saved to profile [kamp]
// for the current session. The SAVE_CONFIG command will
// update the printer config file and restart the printer."
PS:  I have checked, and the nozzle is completely clean.

CChen616, 2024-08-24:
When attempting platform calibration through the screen, the machine will perform infinite downward homing to level the platform as much as possible. Afterward, further adjust the nuts to ensure the PEI sheet is level. Check if this improves the results.

The Q1 Pro bypasses Klipper's Z-offset logic, so modifying related settings in the `printer.cfg` file will not take effect and could even negatively impact leveling.

hoangchieu91, 2024-08-24:
Update from me: I have fixed the G-code. It seems like G-code test_zoffset is better than G-code set_zoffset, or there might be a bug in your code set_zoffset.

> [gcode_macro PRINT_START]
gcode:
    AUTOTUNE_SHAPERS

    {% set bedtemp = params.BED|int %}
    {% set hotendtemp = params.HOTEND|int %}
    {% set chambertemp = params.CHAMBER|default(0)|int %}
    set_zoffset
    {% if chambertemp == 0 %}
        M106 P3 S255
    {% endif %}
    M104 S0
    M140 S{bedtemp}    
    M141 S{chambertemp}
    G28    
    CLEAR_NOZZLE HOTEND={hotendtemp}
    test_zoffset
    M190 S{bedtemp}   
#    Z_TILT_ADJUST
    G29
    G0 Z50 F600
    G0 X0 Y0  F6000
    
    M191 S{chambertemp}
    M109 S{hotendtemp}
    M204 S10000

    ENABLE_ALL_SENSOR
    save_last_file

My printer has now correctly obtained the Z-offset.

TinkerBarn, 2024-08-28:
> > // Probe samples exceed tolerance. Retrying...
> > 20:00:54
Is this message normal? Because, I have often this message if I print with heated chamber with 55°C for ABS and bed temperature with 70°C.
If I print PLA (no chamber heating and  60°C bed temp) this message comes never.
Could it be that either the piezo bed sensors or the inductive (or is it a capacitive?) head sensor have a temp issue?

CChen616, 2024-08-29:
> Is this message normal? Because, I have often this message if I print with heated chamber with 55°C for ABS and bed temperature with 70°C.
> If I print PLA (no chamber heating and 60°C bed temp) this [...]

When leveling the Q1 model, the piezoelectric plate under the heated bed is used. High temperatures can indeed affect its characteristics, leading to changes in sensitivity and, subsequently, a higher likelihood of false readings.

Upgrading to version 4.4.21 can help mitigate this issue to some extent.

If you still find that the error rate is unacceptably high and it impacts your user experience, it might be due to a defective piezoelectric plate (although this is quite rare). In such cases, you can contact customer support to arrange for a replacement.

TinkerBarn, 2024-08-29:
> Upgrading to version 4.4.21 can help mitigate this issue to some extent.

The Q1 is already on 4.4.21

> If you still find that the error rate is unacceptably high and it impacts your user experience, it might be due to a defective piezoelectric plate (although this is quite rare). In such cases, you can contact customer support to arrange for a replacement.

I was already in contact, but got only one piezoelectric plate. But there are 3 plates under the bed. 
How I can find out which one or which ones (could be mire than one) is defective.
Also it is very complex to replace them, because you can’t not see what you do or you need to disassemble the whole bed. Therefore it is in my opinion not possible to do a trial and error with only one replacement piezoelectric plate.
What is your suggestion in this case?

CChen616, 2024-08-29:
You can request additional piezoelectric plates from customer support. 

Currently, there's no better solution for installation and replacement.

JefkeSneltrekker, 2024-11-11:
> Update from me: I have fixed the G-code. It seems like G-code test_zoffset is better than G-code set_zoffset, or there might be a bug in your code set_zoffset.
>
>
> ```
> {% set bedtemp = params.BED|int [...]

Came here to give a big thank you! 
I have a brand new Q1 pro, updated to the latest firmware, and it was also having inconsistent first layers, sometimes to close, sometimes to far away, also ranging from -0.12 to 0.15.
Printer is printing perfectly fine once the first layer is ok.

This macro change fixed all my Z-offset issues on my brand new q1 pro.
I compared the macro's test_zoffset and set_zoffset, but i'm not fully understanding why this change fixes the problem i was having. This is something that maybe should be default implemented by Qidi?

beananimal, 2024-12-26:
I too am at my wits-end with z-offset issues. It is not a dirty nozzle. New printer - was okay for a week. it "just worked". Now I have to babysit first layer and make adjustments on almost every print. from -0.08 to 0.08. 

config.mksini value is sometimes persistent and sometimes not. No rhyme or reason between prints or reboots.  It feels like auto leveling does mesh, but does not account for proper Z, other than the mesh offset. Can Qidi please address the various comment threads with something official other than "clean the nozzle" please.

ProfilerCZ, 2025-01-30:
Thank you, I had the same problem with this printer. When printing PLA, PETG everything was always OK. When printing ASA or ABS the Z-offset was always wrong, it even damaged my bed a few times. I rewrote the G-code according to your parameters and so far everything is OK. Thank you

mikeytheb1, 2025-02-04:
So - I am a complete noob. 
Where is the files you are talking about editing?

ProfilerCZ, 2025-02-05:
> Od: "mikeytheb1" ***@***.***>
> Komu: "QIDITECH/QIDI_Q1_Pro" ***@***.***>

First turn on the printer, then open the slicer. Then proceed according to the photo. Sorry for the translation, I don't speak English.

beananimal, 2025-02-05:
@CChen616 can you please address this issue and others. Reddit, the official Discord, and this very issue section are riddled with people who have to continually adjust z-offset because the printer's automatic offset is not correct. 

In fact, per the discord the actual latest source here does not actually match the firmware on the printer pushed from the update servers. 

It also appears that there has been zero activity on ANY bug fixes or improvements to the Q1  in many months, but the printer is released less than a year ago. Has Qidi abandoned further big fixes or development for this NEW product already?

Gimlir, 2025-05-09:
Same issue there, get_zoffset macro sometimes finishes with wrong offset measuret what results in scratched bed, several times already. In order to fix it tried to move probing point from center of bed to left front corner right above one of sensors - frequency of this issue became a bit lower, but issue is still exist.

Only corellation i can see in all my attempts to catch root cause is temparature, seems piezoelectric sensor is highly dependent from temparature, as all the times issue occurs with hi bed temp (near 100C) and/or enabled chamber heater.

As workarround seems it is ok, at least for me, to skip recalibration of z offset for some prints. 
@QIDITECH please advice how to disable/skip get_zoffset, maybe just for prints with hot bed and chamber, and use previously measured z_offset captured in more  "piezoelectric sensor friendly" environment during previous print.

As long term solution please consider replacement procedure and/or spare part instead of current not reliable solution based on piezoelectric sensors.