QIDITECH/QIDI_Q1_Pro #78: Instructions for updating klipper/moonraker

Ivanuke, 2024-08-23:
I managed to update fluidd to version 1.28.1 to get the print queue functionality that I need. The issue I am having is with qidi version of klipper/moonraker I believe my print queue is terminating after 2-3 print jobs (i have automatic part removal) Print queue works great with my other machines running up to date klipper.

Are there instructions for successfully updating klipper and moonraker specifically on the Q1 Pro? I see there are instructions for the other Qidi X printers but they seem to have different toolboards and I dont want to follow those instructions if they are not the same.

CChen616, 2024-08-24:
To upgrade Klipper, the primary step involves flashing the firmware to the MCU. Since there's no bootloader, the printhead board must be connected via USB to another device for flashing.

For the X 3 series, the printhead board connects through USB, allowing you to flash it by booting the Raspberry Pi and connecting it to the host machine. However, the Q1 uses a different connection method, so direct flashing isn't possible. You'll need to manually remove the printhead board and use a `usb to 4 pin adapter cable`, as shown in the image, to connect it to your PC for flashing.

[image]
[image]

### Make Menuconfig Configuration:

- **Mainboard MCU**: Rename the `klipper.bin` file to `qd_mcu.bin` and flash it to the mainboard via an SD card.
[image]

- **Printhead Board MCU**: Flash the `klipper.uf2` file to the MCU.
[image]

### Moonraker Update:
You can update Moonraker directly to the latest official version, and it should function correctly for normal printing. However, this may impact other features, such as screen thumbnails and the connection to the QIDI Slicer.

Ivanuke, 2024-08-24:
Thank you,

Are there specific files that I need to replace after installing the latest klipper version? My fluidd interface is giving errors such as

- Choice 'submaxmin' for option 'samples_result' in section 'smart_effector' is not a valid choice
- Section 'gcode_shell_command power_loss_resume' is not a valid config section
- Option 'endstop_pin_reverse' is not valid in section 'stepper_z',

CChen616, 2024-08-24:
- Choice 'submaxmin' for option 'samples_result' in section 'smart_effector' is not a valid choice
Associated with the `probe.py` file, the point-sampling method has been improved by eliminating the maximum and minimum values to reduce errors. To adjust, simply change the `samples_result` value in the probe section to `average`.

- Section 'gcode_shell_command power_loss_resume' is not a valid config section
The `gcode_shell_command.py` plugin has been utilized to run the power loss resume shell script.
https://github.com/dw-0/kiauh/blob/master/resources/gcode_shell_command.py

- Option 'endstop_pin_reverse' is not valid in section 'stepper_z',
Linked with the `stepper.py` file, a feature was implemented to enable sensor-less downward homing during platform calibration. You can remove the configuration items with the same prefix but with the `reverse` suffix, though this will cause an error when clicking on platform calibration on the screen, as the corresponding Gcode macro won't be registered.

If you can provide the current Klipper version (commit ID), I can try to sync the updates. However, for the other files and features, I won't be providing a modified version at this time.

Ivanuke, 2024-08-24:
81de9a8615df667ae2ea6d2b0f9204e7f3b09bcc

CChen616, 2024-08-24:
Due to significant changes made to the probe-related files in the Q1 model, and with the recent Klipper version restructuring these files, compatibility with the new version may take considerable time. Therefore, I am only providing the `probe.py`, `qdprobe.py`, `bed_mesh.py`, and `smart_effector.py` files from version 0.10.0.

Other files have been modified based on the version you provided.

These changes have not been tested yet, so there may be some issues.

[klippy.zip](https://github.com/user-attachments/files/16735918/klippy.zip)

dewi-ny-je, 2024-09-04:
> 81de9a is the ID

If successful write a complete guide please

Pukkahq, 2024-09-07:
Try this https://github.com/frap129/OpenQ1

MicroSur, 2024-09-12:
I can't save the webcam in mainsail.
We really need a latest Klipper and Moonraker.

tschreiner, 2024-11-05:
@MicroSur  check https://github.com/frap129/OpenQ1

@Ivanuke  How did you implement the automatic parts removal?

dewi-ny-je, 2024-12-13:
@Ivanuke what's the status? have you updated to latest klipper while maintaining screen functionality?

muzz00ka, 2025-02-04:
> To upgrade Klipper, the primary step involves flashing the firmware to the MCU. Since there's no bootloader, the printhead board must be connected via USB to another device for flashing.
>
> For the X [...]

@CChen616 
Just reading this comment, how can I directly update moonraker? I need 0.8.0+ . I don’t need klipper updated