QIDITECH/QIDI_Q1_Pro #82: Case light takes >5 seconds to turn on/off

wrobelda, 2024-09-05:
As in the title, switching the case light off or on takes about 5 seconds, if not more. This is with most recent firmware, I don't remember this being the case before. Tried using both the printer touch screen and fluidd, it makes no difference. Looking at the log in Fluidd, `SET_PIN PIN=caselight VALUE=1` is issued immediately, it just takes this long for the printer to actually react.

Is this expected?

whb0514, 2024-09-06:
If your machine is executing other commands, the light button will implement a queuing mechanism to delay execution, which is a normal phenomenon.

wrobelda, 2024-09-09:
Isn't there a way to have a command be placed to front of the command queue?

QIDITECH, 2024-09-10:
There is no other way, because klipper's instructions are preloaded and cannot be inserted.

closed 2024-09-10 by QIDITECH