QIDITECH/QIDI_Q1_Pro #83: both of my q1 pro's led strips are flickering badly

Tyler-DurdenRUL1, 2024-09-09:
hello,
    both of my q1 pro printers led light strips flicker badly while printing, they never did so for the first 6 months of owning them. after one of the previous updates is when they started flickering. I saw/read a closed issue concerning the lighting where CChen instructed to insert a command in printer.cfg to help with that. changing stock pmw from 0.1. to 0.0100 . so I did as instructed in the post and it made the flickering much worse. it would flicker pretty badly just sitting in idle with the light on, after I did so. it is driving me nuts and I really want to get this this fixed somehow. any help is much appreciated

QIDITECH, 2024-09-09:
The firmware update does not cause this problem, please contact the after-sales service to try to replace the led strips.

slapsticker75, 2024-09-14:
same at my qidi q1 pro, after updating the firmware today to V4.4.24, the case light starts flickering. until today i got no probs with the case light, it started with the firmware update. any solution from your side?

slapsticker75, 2024-09-20:
> I have no where to look on this setup but I've added lights in the past on my marlin setups. In their cased I'd enable the lights by putting in the command M355 S1 P128. The lights are controlled via PWM (Pulse Width Modulation) so it constantly switches on and off to control the brightness. S0 is off while S1 is on. P1 is the lowest brightness while P255 is full brightness. The dimmer it got the more it flickered. This made dimming the lights pointless in my opinion as I couldn't handle the flickering. My guess is that this is the same issue for this. Somehow the value for the LEDs are set lower than 255 thus the annoying flicker. This can easily be fixed by a single command.
> 
> This is my guess anyhow.

thx for your tip/solution. the q1pro stopped flickering on the next day, i changed nothing. so may be he got a bad day i will observe this furtheron

slapsticker75, 2024-10-05:
> same at my qidi q1 pro, after updating the firmware today to V4.4.24, the case light starts flickering. until today i got no probs with the case light, it started with the firmware update. any solution from your side?

flickering is back

QIDITECH, 2024-10-07:
If the input voltage is unstable, it will also cause the LED light to blink, and we have not made any adjustments to the firmware, so I think this scenario is more likely.

Tyler-DurdenRUL1, 2024-10-07:
> The firmware update does not cause this problem, please contact the after-sales service to try to replace the led strips.

well this issue occurred on both of my Q1 pros at the same exact time, right after the last update. you might not have "done anything with the leds" but you guys might have changed some of the settings around that may draw extra power, causing the flickering. I've even checked them with voltage meter and they both read 23.9v at both end on the strips, sitting idle. as soon as it makes any type of movement, or heats the nozzle or bed it flickers like crazy! both of my q1 pros ( 2 - q1 pro printers ) never had an issue at all, and both started flickering after that update at the same exact time, so it is definitely related in some way to the update. both leds would not go bad at the exact same time. ill have to contact support about it later, right now I am having an issue with trying to get my brand Plus4 printer problem fixed. I received my Plus4 printer 3 days ago and I cant get it to update properly. I have 6 Qidi printers and I always unpack them, set them up, go through the start up process, then immediately update the printers to the newest firmware update via thumb drive as instructed. after one of the previous updates AS well, about 5 months ago, all 6 of my qidi printers get errors now when I update them. it only updates the UI and does not update the SOC for some odd reason on all 6 of my Qidi printers. an error message pops up on the screen saying that the soc is still the older version after trying the initial try for the newest update. I usually can fix it by copying the UI portion of the update after the 1st initial try. then I remove the UI from the thumb drive and then I redo the update and it fixes it. well I cant get my plus4 fixed that way for some reason. I've had it for 3 days now and I cant print anything on it yet. I am getting extremely frustrated that I cant even use my new printer yet. so all my focus is on that right now.

slapsticker75, 2024-10-07:
> If the input voltage is unstable, it will also cause the LED light to blink, and we have not made any adjustments to the firmware, so I think this scenario is more likely.

Then it seems to be a problem with the internal power supply, right? But I don't know why this has occurred since the software update. Something must be different since the update. How can I get a new power supply unit to check this?