QIDITECH/QIDI_Q1_Pro #84: 4.4.24 shows up as 4.4.23 when installed on Q1 Pro

Rizadium, 2024-09-09:
Installed offline update via usb from https://github.com/QIDITECH/QIDI_Q1_Pro/releases/download/Q1_Pro_V4.4.24/Q1_V4.4.24.zip
Q1 Pro shows 4.4.23 was installed instead of 4.4.24.
If i do online check after the update, it will show there is an update to 4.4.24
[image]
[image]

RCsForVetsPREZ, 2024-09-09:
same for me. now it dont want to connect via wifi to orcaslicer......

QIDITECH, 2024-09-11:
Only the version number has a problem, does not affect other use, you can choose not to update to 4.4.24.
We have repaired and replaced the offline installation package.

closed 2024-09-11 by QIDITECH