QIDITECH/QIDI_Q1_Pro #85: How to display message from user macro on printer display?

Deltaflyer7, 2024-09-10:
Ich möchte ein Makro zu meiner gccode_macro.cfg hinzufügen, das am Ende eines Drucks einen Countdown von 15 Minuten startet. Der Countdown startet und dann soll jede Minute die verbleibende Zeit auf dem Druckerdisplay angezeigt werden, unter dem Text benötige ich eine 'Cancel'-Taste zum Abbrechen des Countdowns. Nach Ablauf des Countdowns wird der B-1 5V-Anschluss (Pin PA3) auf der Hauptplatine des Druckers geschaltet, wodurch der Drucker ausgeschaltet wird (Remote-Buchse) Das Makro mit dem Countdown und der Umschaltung funktioniert, aber mein Makro bekommt keine Kommunikation zum Druckerdisplay, um die Meldung anzuzeigen. Ich habe es sowohl mit M117 als auch mit Set_Display_Text msg , versucht, jedoch ohne Erfolg, ich bekomme nur Nachrichten zum Terminal gesendet.  Ich weiß, dass das Display den größten Teil der Benutzeroberfläche verwaltet, was bedeutet, dass alles in der vom Benutzer eingestellten Sprache angezeigt wird, aber ich gehe davon aus, dass Klipper z.B. Fehlermeldungen, die während des Starts auftreten, wie Syntaxfehler in Makros und dergleichen,  nicht in der UI enthalten sind, sondern  direkt als Nachricht aus an das Display sendet, oder eine in der Uienthaltene Message-Box, die externe Messages empfangen kann, aufruft, und den ERROR-Text an diese übergibt.. Wie kann ich also eine Meldung, die auch eine 'Abbrechen'- oder zumindest eine 'OK'-Schaltfläche enthält, an das Display senden und das Drücken der Schaltfläche empfangen, oder een diese in der UI enthaltene Message-Box aufrufen?

MicroSur, 2024-09-13:
M118 may be.

Deltaflyer7, 2024-09-13:
Unfortunately no, M118 only appears in the console / terminal. I tried that first, befor i tried M117. Even SET_DISPLAY_TEXT works not for Text on Printer-Display

whb0514, 2024-09-14:
Sorry, due to the screen being our self-developed system, some display information needs to be adapted to be displayed. Currently, it can only be displayed on the console.

Deltaflyer7, 2024-09-14:
But how can Klipper then show the NAME of the faulty macro on the display in a syntax error message? This name is definitely not part of the display software, as it is a macro I inserted into the gcode_macro. So klipper MUST have access to be able to send specific messages to the display. The attached photo proves that the name of a faulty user macro can be written to the display. so such an interface exists.
[image]

Deltaflyer7, 2024-09-14:
So it cannot be that there is no access to show specific messages on the display that the UI of the display does not know. But it is okay if Qidi does not want to disclose this way, I am working on a solution with an additional hardware module that shows the required information on an additional display, because for me the effort to analyze the complete software is more complex than to directly design a new module with its own display and MCU. The communication between my module and the mainboard and also with the firmware via g-codes is already in progress.

whb0514, 2024-09-14:
Because errors must be displayed to the user and will start with "!!", it is easier to distinguish. However, if all user-defined messages need to be displayed, the logic on the screen will be confusing and require customization.

> So it cannot be that there is no access to show specific messages on the display that the UI of the display does not know. But it is okay if Qidi does not want to disclose this way, I am working on [...]

Deltaflyer7, 2024-09-14:
Yes, this would probably require a message box in the UI, which can be configured via Gcode, with parameters whether a standard 'OK' button or a user button with feedback is required and which can then also accept and display the custom text. Yes, this would require an extension of the UI.