QIDITECH/QIDI_Q1_Pro #86: Camera dissapears after aprinter restart in fluidd webinterface and in the android app

illuminati05, 2024-09-15:
Printer: Qidi Q1 Pro
Firmware: 4.4.21 & 4.4.24

**Problem:** The webcam in fluidd does not work after a printer shutdown and start and the camera settings are defect.

Main GUI View after printer bottup:

[image]

Camera settings corrupt:

[image]

[image]

**Steps to fix:**

1.Enable the camera again:

[image]

You can also delete the camera and add a new one with the same paramters !!!

2.Settings are ok after closing the dialog

[image]

3.Camera works again in Main WebGUI

[image]

**Attached files (LOGs)**

[Logs.zip](https://github.com/user-attachments/files/17006098/Logs.zip)

illuminati05, 2024-09-15:
Printer is 2 weeks old and all stock

illuminati05, 2024-09-17:
A new restart[image]

QIDITECH, 2024-09-18:
There is no problem with the setting of the camera. I think it is caused by the lack of network data in the transmission process.
We will try to replicate and resolve the problem.

illuminati05, 2024-09-18:
Do you need additional log files?

kvdCode, 2024-09-22:
Hello, same problem here !

QIDITECH, 2024-09-24:
We have identified the problem, through the background modification can solve part of the problem, but if you want to completely solve the problem, please wait for the next version of APP and software.

illuminati05, 2024-09-24:
Is it possible to recieve a pre update to test the fix on my printer and if not how long does it take to get a new software version?

QIDITECH, 2024-09-25:
QIDI Link expects to release a new version within two weeks.

illuminati05, 2024-10-10:
Dear guys. How is the progress for the next release to fix this bug

QIDITECH, 2024-10-11:
The new version of the APP has been submitted to the major app stores for approval and can be used after approval.

closed 2024-10-11 by QIDITECH