QIDITECH/QIDI_Q1_Pro #88: Which slicer is dedicated for Q1pro - QidiSlicer or QidiStudio?

derFliegendeHamburger, 2024-09-20:
I have used QidiSlicer up until now. Suddenly with the appearance of the Plus4 printer we see another slicer QidiStudio with support for Q1pro. 
It looks like two competing slicers - what is Qidi's intended strategy for Q1pro owners to use from now on?

QIDITECH, 2024-09-21:
Both software will be perfectly compatible with all QIDI TECH high-speed  3D printers.
In order to cater to customers' habits and hobbies, we have developed both software and will continue to update them at the same time.

derFliegendeHamburger, 2024-09-24:
thank you, but I do not understand.
From the commit logs, you do not yet update both at the same time with the same improvements for Q1pro users.
What benefit  for Q1pro user comes with using either one or the other slicer? 
Maybe you keep one more stable than the other? Which one is (to become) more feature rich?

QIDITECH, 2024-09-25:
According to the actual test results, there may be a few days of time difference between the packaging and release of the two software, but they are basically updated simultaneously and have the same content.
We chose to develop the two software at the same time because the slicing algorithms of the two software have appeared to a certain extent. For some special models and special printing needs, the two software will have different performance, when you encounter this situation, you can have more choices. For the average user, no matter which software you choose, you can get perfect printing, you just choose what you like.

derFliegendeHamburger, 2024-09-25:
thank you for this clarification

closed 2024-09-25 by derFliegendeHamburger