QIDITECH/QIDI_Q1_Pro #90: "!! No trigger on probe after full movement" After 4.24 update.

barrosoman, 2024-09-27:
I update my new printer from 4.16 (I think) to 4.24 and I was having the following message when calibrating the bed: "!! No trigger on probe after full movement". 

To fix it I changed back the line "position_min: -4" in the printer.cfg file on the section [stepper_z].
The older value "position_min: -6" did the trick.

I don't know it this is the only culprit or just a side effect.

QIDITECH, 2024-09-28:
In theory, a firmware update should not cause this problem, but your version span is large, and some unexpected issues may occur. If your modification is valid, please continue to use it, it doesn't matter much. If you want to further eliminate the problem, there are several possibilities, please confirm:
1. Shut down and restart the device
2. The Z-offset value set in the previous version is too large. Clear 0 and reset the value
3. Restore factory Settings

dj0abr, 2025-01-09:
@barrosoman you saved my day. I had the same problem, the trigger point on my machine is at -4.95 so the -4 in printer.cfg must be changed to -6, now all works fine.

ProfilerCZ, 2025-01-30:
Thank you, I had the same problem. After setting according to your post, everything is OK again

Szab06, 2025-02-23:
Hello. Ezt a maerkdown-t hol találom, Előre is köszönöm.

Szab06, 2025-02-23:
> [@barrosoman](https://github.com/barrosoman)megmentetted a napomat. Nekem is ugyanez volt a problémám, a gépemen a trigger pont -4.95-nél van, így a printer.cfg -ben a -4-et -6-ra kell cserélni, most már minden rendben van.

Elnézést. Ezt kifejtené valaki olyan módon,hogy a halandó ember is meg tudja keresni,csinálni a saját gépén? Köszönöm!